ARMmbed/mbed-cli #746: API key provided for `mbed dm` command must be Administrator

dhwalters423, 2018-09-06:
Using the `mbed dm init` command, the API key configured with ` mbed config -G CLOUD_SDK_API_KEY` must be an API key with administrator privileges on Pelion Device Management.

When running the `mbed dm init` command with a 'Developer' privilege level, the following error occurs;

```
$ mbed dm init -d "arm.com" --model-name "f429zi-test"
Traceback (most recent call last):
  File "C:\Users\davwal04\Desktop\workspace\oob\Pelion-ethernet-fw-update\mbed-os\tools\device_management.py", line 155, in <module>
    main()
  File "C:\Users\davwal04\Desktop\workspace\oob\Pelion-ethernet-fw-update\mbed-os\tools\device_management.py", line 150, in main
    }[options.action](options) or 0
  File "C:\Users\davwal04\Desktop\workspace\oob\Pelion-ethernet-fw-update\mbed-os\tools\device_management.py", line 88, in inner
    user = accounts.get_user(api_key.owner_id)
  File "c:\python27\lib\site-packages\mbed_cloud\decorators.py", line 40, in wrapped_f
    raise_(CloudApiException, e, traceback)
  File "c:\python27\lib\site-packages\mbed_cloud\decorators.py", line 33, in wrapped_f
    return fn(*args, **kwargs)
  File "c:\python27\lib\site-packages\mbed_cloud\account_management\account_management.py", line 154, in get_user
    return User(api.get_user(user_id))
  File "c:\python27\lib\site-packages\mbed_cloud\_backends\iam\apis\account_admin_api.py", line 1198, in get_user
    (data) = self.get_user_with_http_info(user_id, **kwargs)
  File "c:\python27\lib\site-packages\mbed_cloud\_backends\iam\apis\account_admin_api.py", line 1274, in get_user_with_http_info
    collection_formats=collection_formats)
  File "c:\python27\lib\site-packages\mbed_cloud\_backends\iam\api_client.py", line 353, in call_api
    _return_http_data_only, collection_formats, _preload_content, _request_timeout)
  File "c:\python27\lib\site-packages\mbed_cloud\_backends\iam\api_client.py", line 113, in wrapped_f
    return fn(self, *args, **kwargs)
  File "c:\python27\lib\site-packages\mbed_cloud\_backends\iam\api_client.py", line 182, in __call_api
    _request_timeout=_request_timeout)
  File "c:\python27\lib\site-packages\mbed_cloud\_backends\iam\api_client.py", line 374, in request
    headers=headers)
  File "c:\python27\lib\site-packages\mbed_cloud\_backends\iam\rest.py", line 235, in GET
    query_params=query_params)
  File "c:\python27\lib\site-packages\mbed_cloud\_backends\iam\rest.py", line 226, in request
    raise ApiException(http_resp=r)
mbed_cloud.exceptions.CloudApiException: (403) "Forbidden" (403)
Reason: Forbidden
HTTP response headers: HTTPHeaderDict({'Content-Length': '270', 'Access-Control-Expose-Headers': 'X-Request-ID,Content-Length,Content-Disposition', 'Set-Cookie': 'TS017f188d=011138c4df3c87d2e9bcac03349923a05fcf2680d4b281a2afac64281925f4333586a8f80063a5a916e0dbf5f48f52e052ee25e487; Path=/', 'Strict-Transport-Security': 'max-age=10886400; includeSubDomains; preload', 'Vary': 'Origin', 'X-Request-ID': '00005b90fba8a6113ab6c9790000000b', 'Connection': 'close', 'Date': 'Thu, 06 Sep 2018 10:04:24 GMT', 'Access-Control-Allow-Origin': 'https://portal.mbedcloud.com', 'Content-Type': 'application/json'})
HTTP response body: {"fields":[{"name":"role","message":"Actor must be an admin or the subject with id 01646b0f25937e8b43d97e8800000000."}],"object":"error","code":403,"type":"access_denied","message":"Not authorized to access the resource.","request_id":"00005b90fba8a6113ab6c9790000000b"}

[mbed] Auto-installing missing Python modules...
[mbed] ERROR: "c:\python27\python.exe" returned error.
       Code: 1
       Path: "C:\Users\davwal04\Desktop\workspace\oob\Pelion-ethernet-fw-update"
       Command: "c:\python27\python.exe -u C:\Users\davwal04\Desktop\workspace\oob\Pelion-ethernet-fw-update\mbed-os\tools\device_management.py init -d arm.com --model-name f429zi-test"
       Tip: You could retry the last command with "-v" flag for verbose output
```

This is a critical issue. Most developers will NOT have access to an Administrator API key.

dhwalters423, 2018-09-06:
To clarify more, the following Groups can be assigned to an API key:

[image]

dhwalters423, 2018-09-06:
Temporary documentation note: https://github.com/ARMmbed/mbed-os-5-docs/pull/710

theotherjimmy, 2018-09-06:
@dhwalters423 I think this may have something do do with the api-key being owned by something other that the current user. The current API should allow you to query for information about the user that owns an API key, which is what the code is doing.

dhwalters423, 2018-09-06:
I think I am misunderstanding.

the workflow is:

1. login to Pelion Portal
2. Create an API key
3. Copy it
4. `mbed config -G CLOUD_SDK_API_KEY`
5. `mbed init...`

There is no way to specify in mbed-cli who owns an API key.

AnotherButler, 2018-09-11:
Does [mbed-os-5-docs PR #710 ](https://github.com/ARMmbed/mbed-os-5-docs/pull/710) solve this issue from a documentation standard for Mbed OS 5.10?

AnotherButler, 2018-09-12:
ping @dhwalters423

oliverjharper, 2018-09-12:
@AnotherButler seems discussion is ongoing about this one, see IOTCORE-416