ARMmbed/mbed-cli #747: Missing help for 'mbed dm' feature

MarceloSalazar, 2018-09-06:
Mbed CLI v1.8.0

The help output should contain more information about the required parameters for device management.

```
$ mbed dm -h

usage: mbed device-management [-h] [-t TOOLCHAIN] [-m TARGET]
                              [--profile PROFILE] [--build BUILD]
                              [--source SOURCE] [-v] [-vv]

Manage Device with Pelion

optional arguments:
  -h, --help            show this help message and exit
  -t TOOLCHAIN, --toolchain TOOLCHAIN
                        Toolchain used for mbed compile
  -m TARGET, --target TARGET
                        Target used for compile for target MCU. Example: K64F,
                        NUCLEO_F401RE, NRF51822...
  --profile PROFILE
  --build BUILD         Build directory. Default: build/
  --source SOURCE       Source directory. Default: . (current dir)
  -v, --verbose         Verbose diagnostic output
  -vv, --very_verbose   Very verbose diagnostic output
```

We'd expect to see information about key parameters/commands, `init`, `domain` and `model-name` 
See docs:
https://github.com/theotherjimmy/Handbook/blob/3d9066d10eef3ad1462b95333284c6a5a31056be/docs/tools/offline/cli-device-management.md

Also applies to: `mbed dm init` and `mbed dm update`

theotherjimmy, 2018-09-06:
@MarceloSalazar This is not the only command with missing help. `mbed compile` and `mbed test` both are missing help.

MarceloSalazar, 2018-09-06:
Help for `mbed compile` is available
```
$ mbed compile -h
usage: mbed compile [-h] [-t TOOLCHAIN] [-m TARGET] [--profile PROFILE]
                    [--library] [--config] [--prefix CONFIG_PREFIX]
                    [--source SOURCE] [--build BUILD] [-c] [-f] [--sterm]
                    [-N ARTIFACT_NAME] [-S [{matrix,toolchains,targets}]]
                    [--app-config APP_CONFIG] [-v] [-vv]
...
```

Help for `mbed tests` is available
```
$ mbed test -h
usage: mbed test [-h] [-t TOOLCHAIN] [-m TARGET] [--compile-list] [--run-list]
                 [--compile] [--run] [-n TESTS_BY_NAME] [--source SOURCE]
                 [--build BUILD] [--profile PROFILE] [-c]
                 [--test-spec TEST_SPEC] [--app-config APP_CONFIG]
                 [--test-config TEST_CONFIG] [--coverage {html,xml,both}]
                 [--make-program {gmake,make,mingw32-make,ninja}]
                 [--generator {Unix Makefiles,MinGW Makefiles,Ninja}]
                 [--new FILEPATH] [-r REGEX] [--unittests]
                 [--build-data BUILD_DATA] [--greentea] [--icetea] [-v] [-vv]
...
```

Help for `mbed dm`, it's missing the `init` subcommand
```
$ mbed dm -h
usage: mbed device-management [-h] [-t TOOLCHAIN] [-m TARGET]
                              [--profile PROFILE] [--build BUILD]
                              [--source SOURCE] [-v] [-vv]
```

Help for `mbed dm init`, it's missing the `-d` and `--model-name` parameters
```
$ mbed dm init -h
usage: mbed device-management [-h] [-t TOOLCHAIN] [-m TARGET]
                              [--profile PROFILE] [--build BUILD]
                              [--source SOURCE] [-v] [-vv]
...
```

Help for `mbed dm update`, it's missing the `device` and `prepare` subcommand and parameters
```
mbed dm update -h
usage: mbed device-management [-h] [-t TOOLCHAIN] [-m TARGET]
                              [--profile PROFILE] [--build BUILD]
                              [--source SOURCE] [-v] [-vv]
...
```

theotherjimmy, 2018-09-06:
@MarceloSalazar 
`mbed compile` runs `make.py`

`make.py`'s help:
```
usage: make.py [-h] [-m MCU] [-t TOOLCHAIN] [--color [-c]
               [--profile PROFILE] [--app-config APP_CONFIG] [-p PROGRAM]
               [-n PROGRAM] [-j JOBS] [-v] [--silent] [-D MACROS]
               [-S [{matrix,toolchains,targets}]] [-f GENERAL_FILTER_REGEX]
               [--stats-depth STATS_DEPTH] [--automated] [--host HOST_TEST]
               [--extra EXTRA] [--peripherals PERIPHERALS]
               [--dep DEPENDENCIES] [--source SOURCE_DIR]
               [--duration DURATION] [--build BUILD_DIR] [-N ARTIFACT_NAME]
               [--ignore IGNORE] [-d DISK] [-s SERIAL] [-b BAUD] [-L] [--rpc]
               [--usb] [--dsp] [--testlib] [--build-data BUILD_DATA]
               [-l LINKER_SCRIPT]
```

missing arguments that a user might actually use: `--color` `-D` `-l` `--ignore` `-j`

-----------------------

My point is that the help is always behind. I think this is an issue with the current architecture of Mbed CLI.

We can improve the help for `mbed dm`.

teetak01, 2018-09-07:
@theotherjimmy the DM is giving out incorrect instructions (effectively those what manifest-tool uses), but which will not work with mbed-cli.

```
usage: device_management.py init [-h] [-c CERTIFICATE] [-k PRIVATE_KEY]
                                 (-V VENDOR_ID | -d VENDOR_DOMAIN)
                                 (-C CLASS_ID | -m MODEL_NAME)
                                 [-S SERVER_ADDRESS] [-a API_KEY] [-q] [-f]
                                 [--psk]
device_management.py init: error: one of the arguments -C/--class-id -m/--model-name is required
```

It is suggesting that I should use "-m" for MODEL_NAME, which is correct for manifest-tool, but conflicts with "--mcu" flag in mbed-cli.

theotherjimmy, 2018-09-07:
@teetak01 Yep. that's also true.

theotherjimmy, 2018-09-07:
To summarize:

* help is incorrect.
* help is incomplete.
* help is missing subcommands.
* `-m` conflict is not resolved.