ARMmbed/mbed-cli #749: mbed-cli always doing some Auto-installing which can be slow

teetak01, 2018-09-07:
With mbed-cli 1.8.0 I always get this trace:

[mbed] Auto-installing missing Python modules...

I have installed the latest mbed-os/requirements.txt files so there should be no such step needed.

theotherjimmy, 2018-09-07:
The "check that you have all the deps" thing may not be perfect. I see this on my mac, but not my Linux box. Weird...

theotherjimmy, 2018-09-07:
What's your system like?
 * OS
 * Python version

teetak01, 2018-09-10:
Python 2.7.12, Ubuntu 16.04

maclobdell, 2018-09-10:
I see this same issue on Windows.  Every time I issue a compile command it results in this, even though I also have the requirements all installed.

```
[mbed] Auto-installing missing Python modules...
```

SeppoTakalo, 2018-09-13:
Autoinstalling also break some Pip environments, as reported here: https://github.com/ARMmbed/mbed-cli/issues/754

screamerbg, 2018-09-13:
Fix was released in Mbed CLI 1.8.1. Could you update and try to reproduce the issue?

mtvg, 2018-11-09:
Also having the issue on Mac (10.13.6) with mbed-cli 1.8.2, Python 2.7.15, pip 18.0

mtvg, 2018-11-12:
Works for me with 1.8.3. Thanks!

Ronny-Liu, 2018-11-15:
Works for me with 1.8.3. Thanks!

korjaa, 2019-01-08:
I had custom icetea installed on my machine with:
```
cd icetea
python setup.py develop
```
now that I ran tried to run `mbed compile` it auto-installed it's own icetea on top of the development version and now neither one of them work.

```
$ mbed --version
1.8.3
```

korjaa, 2019-01-08:
This auto-installation is very frustrating feature.

korjaa, 2019-01-08:
Seems to relate to https://github.com/ARMmbed/mbed-cli/issues/795

SeppoTakalo, 2019-01-08:
’RM -rf mbes-os/requirements.txt’ should help.

bridadan, 2019-01-08:
I would like to remove the need to install these requirements in the future. However, this is difficult to do while the Mbed OS build system is still present alongside the source code.

loverdeg-ep, 2019-01-08:
>  while the Mbed OS build system is still present alongside the source code.

Tell me more.

screamerbg, 2019-01-08:
@bridadan There will always be some instalation of modules regardless whether the build system is part of the os or as an external tools package. Unless we're taking about integrating and packaging icetea and any other python dependency together in one large tools package for the os.

theotherjimmy, 2019-01-08:
@screamerbg If the build system and other python scripts are no longer part of the OS, then the requirements.txt will move here (into Mbed CLI) and dependencies would be installed up front.

flit, 2019-02-26:
I was seeing this bug prior to 1.8.3 on macOS 10.14.3. After updating to 1.8.3, I can confirm that it's fixed.

ItsNageeK, 2019-08-20:
Having this issue, mbed 1.10.0, macOS 10.14.4, python 2.7.14, pip 19.2.2, `pip install -r requirements.txt` continues to display the "missing Python modules..." any suggestions?

mark-edgeworth, 2019-08-20:
We think this issue will be resolved with the next release of the mbed-cli tools package which were are working on now.

adbridge, 2019-11-01:
Apologies, this was closed in error by the automated bot - re-opening.

mark-edgeworth, 2019-12-04:
No further work planned on this.

closed 2019-12-04 by mark-edgeworth