ARMmbed/mbed-cli #754: Mbed-cli install Pip packages

SeppoTakalo, 2018-09-13:
mbed-cli install pip packages and by doing so might break current environment.

```
$ pip install -U jsonschema==2.4.0
$ pip list |grep jsonschema
jsonschema                    2.4.0
$ mbed compile --config
[mbed] Auto-installing missing Python modules...

...
$ pip list |grep jsonschema
jsonschema                    2.6.0
```

Installing Pip packages might have unwanted side effect and should not be done if user is not requesting that.
Such feature should be removed from mbed-cli

This blocks Icetea testing on my Mac.

## Example:

```
$ icetea --version
1.0.1

# Run any mbed command
$ mbed compile --config >/dev/null
$ icetea --version
Traceback (most recent call last):
  File "/usr/local/bin/icetea", line 11, in <module>
    sys.exit(icetea_main())
  File "/usr/local/lib/python2.7/site-packages/icetea_lib/main.py", line 22, in icetea_main
    returnCode = manager.run()
  File "/usr/local/lib/python2.7/site-packages/icetea_lib/IceteaManager.py", line 219, in run
    version = get_fw_version()
  File "/usr/local/lib/python2.7/site-packages/icetea_lib/tools/tools.py", line 51, in get_fw_version
    pkg = require(get_fw_name())[0]
  File "/usr/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 892, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 783, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (jsonschema 2.6.0 (/usr/local/lib/python2.7/site-packages), Requirement.parse('jsonschema<=2.4.0'), set(['jsonmerge']))
```

## Conclusion

Running any Mbed command breaks Icetea

SeppoTakalo, 2018-09-13:
CC: @jupe @OPpuolitaival @jonikula

screamerbg, 2018-09-13:
Mbed CLI is not installing packages on it's own. Mbed CLI asks pip to check and respect the requirements.txt file inside Mbed OS. When requirements.txt is referencing a python module that requires new version of a module (like jsonschema), then new version will be installed to respect the python module requirement.

E.g. either bind icetea to specific version of it's python requirements, or fix the conflicting requirement from other python requirement in requirements.txt. Don't blame Mbed CLI for it.

SeppoTakalo, 2018-09-13:
> Mbed CLI is not installing packages on it's own. Mbed CLI asks pip to check and respect the requirements.txt file inside Mbed OS

Your second sentence contradicts the first one.

Mbed CLI is installing packages. Period.
It happens here:
https://github.com/ARMmbed/mbed-cli/blob/master/mbed/mbed.py#L1622

Mbed OS is not asking this to happen.
Mbed OS supplies list of requirements, which might or might not, be supplied by my environment.
This list might, or might not, conflicts other test tools in my current environment. That is a separate issue.

I did not ask `mbed` to update.

You cannot close this issue, just because Mbed OS requirements, and Icetea requirements are conflicting. I need to report that as a separate issue.

theotherjimmy, 2018-09-13:
@screamerbg At no point did @SeppoTakalo blame Mbed CLI for the conflicts. He just blamed it for doing something that he did not ask for: installing the requirements.

theotherjimmy, 2018-09-13:
@SeppoTakalo Does the switch `--no-requirements` resolve this? or would you need something more?

theotherjimmy, 2018-09-13:
Upon further inspection of this issue the "breaking of pip environment" is done by the fact that icetea depends on a specific version of `jsonschema` that is:
 * not documented, or specified
 * in conflict with the requirements of the mbed-os build tools specified requirements.

This issue should move to icetea. As such, the issue as written was correctly marked as invalid and closed by @screamerbg. I made a mistake in re-opening it, and I will be repeating those steps.

---------------------

The other thing we were discussing is auto-installing and if it should happen. The big wrinkle is that Mbed CLI calls out to Mbed OS build tools, and is responsible for making sure that they have all of their requirements. The following subcommands could update Mbed OS, which would change the build tools requirements. Therefore the following subcommands _should_ install requirements:
 * `mbed new`
 * `mbed import`
 * `mbed add`
 * `mbed deploy`
 * `mbed update`

The following subcommands are not supposed to interact with the network, unless specified, and currently might access the network through pip:
 * `mbed compile`
 * `mbed test`
 * `mbed export`

I'm going to open another issue to address this.

closed 2018-09-13 by theotherjimmy