ARMmbed/mbed-cli #756: `mbed compile` can access the network, through pip

theotherjimmy, 2018-09-13:
Mbed CLI calls out to Mbed OS build tools delivered with Mbed OS, and is responsible for making sure that they have all of their requirements. The following subcommands could update Mbed OS, which would change the build tools requirements. Therefore the following subcommands _should_ install requirements:
 * `mbed new`
 * `mbed import`
 * `mbed add`
 * `mbed deploy`
 * `mbed update`

The following subcommands are not supposed to interact with the network to allow offline operation, and currently might access the network through pip:
 * `mbed compile`
 * `mbed test`
 * `mbed export`

These commands should not auto-install, so that offline work is not interrupted.

@screamerbg What do you think?

screamerbg, 2018-09-13:
@theotherjimmy Agreed. Thanks for highlighting this. I'll issue a PR later today to address this.

theotherjimmy, 2019-02-11:
#833 Resolved this.

closed 2019-02-11 by theotherjimmy