ARMmbed/mbed-cli #759: Compile fails for long arg list in build_profile

mbartling, 2018-09-17:
Almost exactly the same issue as https://github.com/ARMmbed/mbed-cli/issues/339

- mbed-cli: 1.8.1
- arm-none-eabi-g++ (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
- macOS High Sierra 10.13.6

```
micbar02 @ C02SP20YFVH4 ~/git/tf/rom-demo/utensor-mnist-demo (f/rom-tensor *)
└─ $ ▶ mbed compile -vv --profile=utensor/build_profile/release.json
...
[mbed-24823] Exec "/Users/micbar02/anaconda2/bin/python -u /Users/micbar02/git/tf/rom-demo/utensor-mnist-demo/mbed-os/tools/make.py -t gcc_arm -m DISCO_F413ZH --profile utensor/build_profile/release.json --source . --build ./BUILD/DISCO_F413ZH/GCC_ARM-RELEASE -v" in "/Users/micbar02/git/tf/rom-demo/utensor-mnist-demo"
Building project utensor-mnist-demo (DISCO_F413ZH, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
[DEBUG] Macros: -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -D__FPU_PRESENT=1 -DDEVICE_PORTOUT=1 -DUSBHOST_OTHER -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -D__CMSIS_RTOS -DDEVICE_LOWPOWERTIMER=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -DTARGET_STM32F4 -DTARGET_DISCO_F413ZH -DTOOLCHAIN_GCC -DDEVICE_CAN=1 -DTARGET_CORTEX_M -DDEVICE_I2C_ASYNCH=1 -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DDEVICE_ANALOGIN=1 -DDEVICE_SPI_ASYNCH=1 -DDEVICE_PWMOUT=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_FAMILY_STM32 -DTARGET_FF_ARDUINO -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_STM -DDEVICE_SERIAL_FC=1 -DTARGET_STM32F413xH -DDEVICE_TRNG=1 -DTARGET_LIKE_MBED -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DTARGET_STM32F413xx -DDEVICE_SPI=1 -DUSB_STM_HAL -DTARGET_STM32F413ZH -DDEVICE_SPISLAVE=1 -DMBED_BUILD_TIMESTAMP=1537200319.87 -DDEVICE_SERIAL=1 -DDEVICE_FLASH=1 -DARM_MATH_CM4
Compile [  0.2%]: ft6x06.c
[DEBUG] Compile: arm-none-eabi-gcc -std=gnu99 -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -DNDEBUG -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -D__FPU_PRESENT=1 -DDEVICE_PORTOUT=1 -DUSBHOST_OTHER -DDEVICE_PORTINOUT=1 -DTARGET_RTOS_M4_M7 -D__CMSIS_RTOS -DDEVICE_LOWPOWERTIMER=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -DTARGET_STM32F4 -DTARGET_DISCO_F413ZH -DTOOLCHAIN_GCC -DDEVICE_CAN=1 -DTARGET_CORTEX_M -DDEVICE_I2C_ASYNCH=1 -DTARGET_LIKE_CORTEX_M4 -DDEVICE_ANALOGOUT=1 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DDEVICE_ANALOGIN=1 -DDEVICE_SPI_ASYNCH=1 -DDEVICE_PWMOUT=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_FAMILY_STM32 -DTARGET_FF_ARDUINO -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_STM -DDEVICE_SERIAL_FC=1 -DTARGET_STM32F413xH -DDEVICE_TRNG=1 -DTARGET_LIKE_MBED -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DTARGET_STM32F413xx -DDEVICE_SPI=1 -DUSB_STM_HAL -DTARGET_STM32F413ZH -DDEVICE_SPISLAVE=1 -DMBED_BUILD_TIMESTAMP=1537200319.87 -DDEVICE_SERIAL=1 -DDEVICE_FLASH=1 -DARM_MATH_CM4 @./BUILD/DISCO_F413ZH/GCC_ARM-RELEASE/.includes_7d2d57ac3ad0714f93c9bb5b82d4592c.txt -include ./BUILD/DISCO_F413ZH/GCC_ARM-RELEASE/mbed_config.h -MD -MF BUILD/DISCO_F413ZH/GCC_ARM-RELEASE/BSP_DISCO_F413ZH/Drivers/BSP/Components/ft6x06/ft6x06.d -o BUILD/DISCO_F413ZH/GCC_ARM-RELEASE/BSP_DISCO_F413ZH/Drivers/BSP/Components/ft6x06/ft6x06.o ./BSP_DISCO_F413ZH/Drivers/BSP/Components/ft6x06/ft6x06.c
[DEBUG] Return: 1
[DEBUG] Output: arm-none-eabi-gcc: error trying to exec '/Users/micbar02/gcc/gcc-arm-none-eabi-7-2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/cc1': execv: Argument list too long
Traceback (most recent call last):
  File "/Users/micbar02/git/tf/rom-demo/utensor-mnist-demo/mbed-os/tools/make.py", line 296, in <module>
    stats_depth=options.stats_depth)
  File "/Users/micbar02/git/tf/rom-demo/utensor-mnist-demo/mbed-os/tools/build_api.py", line 533, in build_project
    objects = toolchain.compile_sources(resources, resources.inc_dirs)
  File "/Users/micbar02/git/tf/rom-demo/utensor-mnist-demo/mbed-os/tools/toolchains/__init__.py", line 928, in compile_sources
    return self.compile_queue(queue, objects)
  File "/Users/micbar02/git/tf/rom-demo/utensor-mnist-demo/mbed-os/tools/toolchains/__init__.py", line 991, in compile_queue
    raise ToolException(err)
ToolException: arm-none-eabi-gcc: error trying to exec '/Users/micbar02/gcc/gcc-arm-none-eabi-7-2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/cc1': execv: Argument list too long
```

mbartling, 2018-09-17:
FYI, reproducible on GCC 6.

```
"-I." "-I./BSP_DISCO_F413ZH" "-I./BSP_DISCO_F413ZH/Drivers" "-I./BSP_DISCO_F413ZH/Drivers/BSP" "-I./BSP_DISCO_F413ZH/Drivers/BSP/Components" "-I./BSP_DISCO_F413ZH/Drivers/BSP/Components/Common" "-I./BSP_DISCO_F413ZH/Drivers/BSP/Components/ft6x06" "-I./BSP_DISCO_F413ZH/Drivers/BSP/Components/n25q128a" "-I./BSP_DISCO_F413ZH/Drivers/BSP/Components/st7789h2" "-I./BSP_DISCO_F413ZH/Drivers/BSP/Components/wm8994" "-I./BSP_DISCO_F413ZH/Drivers/BSP/STM32F413H-Discovery" "-I./BSP_DISCO_F413ZH/Utilities" "-I./BSP_DISCO_F413ZH/Utilities/Fonts" "-I./F413ZH_SD_BlockDevice" "-I./constants" "-I./constants/deep_mlp" "-I./docs" "-I./docs/images" "-I./mbed-os" "-I./mbed-os/cmsis" "-I./mbed-os/cmsis/TARGET_CORTEX_M" "-I./mbed-os/drivers" "-I./mbed-os/events" "-I./mbed-os/events/equeue" "-I./mbed-os/features" "-I./mbed-os/features/filesystem" "-I./mbed-os/features/filesystem/bd" "-I./mbed-os/features/filesystem/fat" "-I./mbed-os/features/filesystem/fat/ChaN" "-I./mbed-os/features/filesystem/littlefs" "-I./mbed-os/features/filesystem/littlefs/TESTS_COMMON" "-I./mbed-os/features/filesystem/littlefs/littlefs" "-I./mbed-os/features/frameworks" "-I./mbed-os/features/frameworks/greentea-client" "-I./mbed-os/features/frameworks/greentea-client/greentea-client" "-I./mbed-os/features/frameworks/greentea-client/source" "-I./mbed-os/features/frameworks/unity" "-I./mbed-os/features/frameworks/unity/source" "-I./mbed-os/features/frameworks/unity/unity" "-I./mbed-os/features/frameworks/utest" "-I./mbed-os/features/frameworks/utest/source" "-I./mbed-os/features/frameworks/utest/utest" "-I./mbed-os/features/mbedtls" "-I./mbed-os/features/mbedtls/importer" "-I./mbed-os/features/mbedtls/inc" "-I./mbed-os/features/mbedtls/inc/mbedtls" "-I./mbed-os/features/mbedtls/platform" "-I./mbed-os/features/mbedtls/platform/inc" "-I./mbed-os/features/mbedtls/platform/src" "-I./mbed-os/features/mbedtls/src" "-I./mbed-os/features/mbedtls/targets" "-I./mbed-os/features/mbedtls/targets/TARGET_STM" "-I./mbed-os/features/mbedtls/targets/TARGET_STM/TARGET_STM32F4" "-I./mbed-os/features/nanostack" "-I./mbed-os/features/netsocket" "-I./mbed-os/features/netsocket/cellular" "-I./mbed-os/features/netsocket/cellular/generic_modem_driver" "-I./mbed-os/features/netsocket/cellular/utils" "-I./mbed-os/features/storage" "-I./mbed-os/hal" "-I./mbed-os/hal/storage_abstraction" "-I./mbed-os/platform" "-I./mbed-os/rtos" "-I./mbed-os/rtos/TARGET_CORTEX" "-I./mbed-os/rtos/TARGET_CORTEX/rtx4" "-I./mbed-os/rtos/TARGET_CORTEX/rtx5" "-I./mbed-os/rtos/TARGET_CORTEX/rtx5/Include" "-I./mbed-os/rtos/TARGET_CORTEX/rtx5/RTX" "-I./mbed-os/rtos/TARGET_CORTEX/rtx5/RTX/Config" "-I./mbed-os/rtos/TARGET_CORTEX/rtx5/RTX/Include" "-I./mbed-os/rtos/TARGET_CORTEX/rtx5/RTX/Source" "-I./mbed-os/rtos/TARGET_CORTEX/rtx5/RTX/Source/TOOLCHAIN_GCC" "-I./mbed-os/rtos/TARGET_CORTEX/rtx5/RTX/Source/TOOLCHAIN_GCC/TARGET_RTOS_M4_M7" "-I./mbed-os/rtos/TARGET_CORTEX/rtx5/Source" "-I./mbed-os/targets" "-I./mbed-os/targets/TARGET_STM" "-I./mbed-os/targets/TARGET_STM/TARGET_STM32F4" "-I./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F413xH" "-I./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F413xH/TARGET_DISCO_F413ZH" "-I./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F413xH/device" "-I./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F413xH/device/TOOLCHAIN_GCC_ARM" "-I./mbed-os/targets/TARGET_STM/TARGET_STM32F4/device" "-I./models" "-I./tensorflow-models" "-I./tensorflow-models/chkps" "-I./tensorflow-models/mnist_data" "-I./tensorflow-models/mnist_model" "-I./uTensor" "-I./uTensor/build_profile" "-I./uTensor/docs" "-I./uTensor/docs/img" "-I./uTensor/uTensor" "-I./uTensor/uTensor/core" "-I./uTensor/uTensor/loaders" "-I./uTensor/uTensor/ops" "-I./uTensor/uTensor/util" "-I./uTensor/unity" "-I./venv" "-I./venv/bin" "-I./venv/etc" "-I./venv/etc/jupyter" "-I./venv/etc/jupyter/nbconfig" "-I./venv/etc/jupyter/nbconfig/notebook.d" "-I./venv/include" "-I./venv/include/python3.6m" "-I./venv/include/python3.6m/greenlet" "-I./venv/lib" "-I./venv/lib/python3.6" "-I./venv/lib/python3.6/__pycache__" "-I./venv/lib/python3.6/collections" "-I./venv/lib/python3.6/collections/__pycache__" "-I./venv/lib/python3.6/config-3.6m-darwin" "-I./venv/lib/python3.6/config-3.6m-darwin/__pycache__" "-I./venv/lib/python3.6/distutils" "-I./venv/lib/python3.6/distutils/__pycache__" "-I./venv/lib/python3.6/encodings" "-I./venv/lib/python3.6/encodings/__pycache__" "-I./venv/lib/python3.6/importlib" "-I./venv/lib/python3.6/importlib/__pycache__" "-I./venv/lib/python3.6/lib-dynload" "-I./venv/lib/python3.6/site-packages" "-I./venv/lib/python3.6/site-packages/IPython" "-I./venv/lib/python3.6/site-packages/IPython/__pycache__" "-I./venv/lib/python3.6/site-packages/IPython/core" "-I./venv/lib/python3.6/site-packages/IPython/core/__pycache__" "-I./venv/lib/python3.6/site-packages/IPython/core/magics" "-I./venv/lib/python3.6/site-packages/IPython/core/magics/__pycache__" "-I./venv/lib/python3.6/site-packages/IPython/core/profile" "-I./venv/lib/python3.6/site-packages/IPython/core/tests" "-I./venv/lib/python3.6/site-packages/IPython/core/tests/__pycache__" "-I./venv/lib/python3.6/site-packages/IPython/core/tests/daft_extension" "-I./venv/lib/python3.6/site-packages/IPython/core/tests/daft_extension/__pycache__" "-I./venv/lib/python3.6/site-packages/IPython/extensions" "-I./venv/lib/python3.6/site-packages/IPython/extensions/__pycache__" "-I./venv/lib/python3.6/site-packages/IPython/extensi [... cut ...] 1" "-I./venv/lib/python3.6/site-packages/tensorflow/include/tensorflow/include/external/eigen_archive/unsupported/Eigen/CXX11/src" "-I./venv/lib/python3.6/site-packages/tensorflow/include/tensorflow/include/external/eigen_archive/unsupported/Eigen/CXX11/src/Tensor" "-I./venv/lib/python3.6/site-packages/tensorflow/include/tensorflow/include/external/eigen_archive/unsupported/Eigen/CXX11/src/TensorSymmetry" "-I./venv/lib/python3.6/site-packages/tensorflow/include/tensorflow/include/external/eigen_

cmonr, 2018-09-17:
@mbartling I'm confused. Are you including your entire OS?

That's a whole lot python packages in the include list.

mbartling, 2018-09-17:
@cmonr DAMMIT, I've had issues similar to this in the past. It's caused by having a Python virtualenv in my project repo. Closing the ticket

closed 2018-09-17 by mbartling

cmonr, 2018-09-17:
@mbartling Even though this isn't what we thought, it's still a significant-enough issue worth tracking.

Would you mind opening another issue for others to reproduce and so that we can fix?