ARMmbed/mbed-cli #768: Erasing flash exclusively or before flash

nmaiti, 2018-10-05:
I couldn't  locate any erase partition/flash use case in mbed-cli. I may be missing it but I see NVstored partitions are not getting erased during reflash using -f command.

theotherjimmy, 2018-10-05:
@nmaiti reflash's behavoir is 1) not controlled by Mbed CLI and is controlled by the debug circuit and 2) generally depends on the file extension: .bin erases all flash and .hex erases only what is specified.

As we don't currently have any understanding of a "partition" or "NVStore" in the build tools or mbed CLI, I don't see how there is much we can do here.

nmaiti, 2018-10-05:
mbed-cli compile -f option not erasing full flash, as its writing binary to flash, but keeping old nvstore data. Is it something unexpected behavior? using st F32M757 board.

theotherjimmy, 2018-10-08:
@nmaiti I'm not sure what exactly you're asking about. `mbed compile -f` copies the binary to the device. At that point it's implementation defined what that binary does.

evedon, 2019-05-24:
Considering the question has answered so closing this issue.

closed 2019-05-24 by evedon