ARMmbed/mbed-cli #769: Upload to a custom boards

masoudr, 2018-10-18:
Hi,
I'm using an STM32F4 custom board. Is there any option that I can upload the compiled project after `mbed compile` to my board through CLI. Custom boards don't support drag and drop programming, so it should be programmed using external programmer software. It would be good if you add a custom upload option to CLI.
Thanks.

theotherjimmy, 2018-10-18:
Hi @masoudr,

You're correct: we don't currently have an option to use custom programming software. If it's a somewhat standard interface firmware configuration, perhaps we could integrate support for it into the programming/flashing implementation (currently in htrun).

masoudr, 2018-10-18:
@theotherjimmy Currently there are some platforms like `platformio` that has this option. I think CLI can be easily integrated with some tools like `openocd`.

theotherjimmy, 2018-10-19:
@masoudr That's a good point: htrun (the implementation of -f) has a pyocd plugin for both flash and reset. Perhaps this would be sufficient for your use case?

masoudr, 2018-10-19:
@theotherjimmy Actually I haven't tested `htrun` but it seems good. Does it have the ability to upload with both external programmer and serial port (dfu)? It would be better to integrate it with `mbed compile`. Currently, I'm using [this method](https://os.mbed.com/docs/v5.9/tutorials/visual-studio-code.html) to debug but there is no method for only uploading.

theotherjimmy, 2018-10-19:
> . Does it have the ability to upload with both external programmer and serial port (dfu)?

Not that I'm aware of

> It would be better to integrate it with `mbed compile`.

`mbed compile` imports htrun to flash, so anything that htrun can flash, `mbed compile` can flash.

closed 2019-10-31 by ciarmcom