ARMmbed/mbed-cli #770: WARNING: Cannot run icetea tests. Current folder does not contain TEST_APPS folder.

panuniverse, 2018-10-24:
What's the reason?

theotherjimmy, 2018-10-24:
You're using new mbed CLI, that supports an `--icetea` flag with an old version of Mbed OS. Perhaps that should be an error if you use `--icetea`

panuniverse, 2018-10-26:
[image]

panuniverse, 2018-10-26:
[image]

I have resolved that.
Because of the mbed_cli version, need to switch version.

theotherjimmy, 2018-10-26:
@panuniverse The warning is a red herring: the reason it stopped is because the filesystem_recovery/resilience_functional test failed to compile.

maclobdell, 2018-10-29:
I recently noticed this warning also.  I don't agree that it needs to indiscriminately warn users that they 'Cannot run icetea tests.' Especially, since most people will not be running Ice Tea tests.   They likely will not know what an Ice Tea test even is.  This warning might send them down a rabbit hole (e.g. this ticket).  Is it possible to suppress the warning?  To me, it is useless.  :)

panuniverse, 2018-10-30:
@theotherjimmy Thank you,  you are right.

panuniverse, 2018-10-30:
But the mbed_cli version is one reason which causes the Error.

elm3, 2018-11-12:
Is this now fixed?  We closed the warning issue, and the original poster has upgrade CLI version to fix.

naveenkaje, 2019-03-13:
With https://github.com/ARMmbed/mbed-cli/pull/776/commits, the WARNING does not show up. Issue can be closed.

closed 2019-03-13 by theotherjimmy