ARMmbed/mbed-cli #771: cc1.exe: error: output filename specified twice

vrushalibhokare, 2018-10-24:
i am using mbed os 5 for compilation, i am getting this error (cc1.exe: error: output filename specified twice) please some one help.

theotherjimmy, 2018-10-24:
>"mfpu=fpv4-sp-d16"

You're missing a `-`

vrushalibhokare, 2018-10-25:
when i put - in command ,we get this VFP  register error and ld returned 1 exit status.
 please any one help me.

[ERROR] c:/users/admin/gnu tools arm embedded/6 2017-q2-update/bin/../lib/gcc/ar
m-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld.exe: error: ./STM32_Audio/Add
ons/PDM/Lib\libPDMFilter_CM4_GCC.a(pdm_filter.o) uses VFP register arguments, ./
BUILD/DISCO_F469NI/GCC_ARM-RELEASE_O3/kws_realtime_test.elf does not
c:/users/admin/gnu tools arm embedded/6 2017-q2-update/bin/../lib/gcc/arm-none-e
abi/6.3.1/../../../../arm-none-eabi/bin/ld.exe: failed to merge target specific
data of file ./STM32_Audio/Addons/PDM/Lib\libPDMFilter_CM4_GCC.a(pdm_filter.o)
collect2.exe: error: ld returned 1 exit status

theotherjimmy, 2018-10-25:
@vrushalibhokare That indicates that you have compiled with `-mfloat-abi=hard` for that library. Try adding `"-mfloat-abi=softfp"` (this still enables hardware floating point, with the calling convention of software emulation).

theotherjimmy, 2018-10-25:
@vrushalibhokare General github tip: add ` ``` ` before and after code blocks to format them correctly. I edited your comments to use them. After the top ` ``` ` you can also add a language name to get syntax highlighting for that language. I used ` ```json ` to get JSON syntax highlighting in your issue description.

vrushalibhokare, 2018-10-26:
I was try this solution but still i getting same error.  please find attached image of this error and help me for this error.

theotherjimmy, 2018-10-26:
@vrushalibhokare There is no attached image.

vrushalibhokare, 2018-10-27:
I was try this solution but still i getting same error. please find attached image of this error and help me for this error.
[image]

vrushalibhokare, 2018-10-27:
please give me solution of this above error.

theotherjimmy, 2018-11-09:
@vrushalibhokare Recompile the library you're using to use `-mfloat-abi=hard`.

theotherjimmy, 2018-11-14:
@vrushalibhokare I think we're going about this in a way that is less helpful to both of us:

>please give me solution of this above error.

Please don't ask me to do your homework. Further, the error you're seeing is expected and correct behavior. Finally, I don't know what you're trying to do (so far you have just said "fix it" with compiler erros). Without context, I can only help you so much.

I understand that asking a good question to help everyone involved is a skill that takes practice. I'm going to try guessing what questions you could ask that are more helpful to both of us. I don't know how much you know (pro tip: include what you have tried and what information you know in the issue/question description)

Perhaps you meant to ask the question:
> What can cause this error?

That's a question that I can help you with. The linker is claiming that the object files contained within the archive, `libPDMFilter_CM4_GCC.a`, are compiled to use "VFP register arguments" (more on this in a bit) and that the executable you're generating, `kws_realtime_test.elf`, does not use them. "VFP register arguments" is a C/C++ calling convention term made from:
 * "VFP" meaning "Vector Floating Point co-processor".
 * "register arguments" are how C/C++ functions pass arguments/parameters to each other in the Arm EABI (Extended Application Binary Interface or machine code used to call functions) calling convention: through registers.

So "VFP register arguments" is referring to how to treat Floating point function call arguments. Specifically using "VFP register arguments" indicates that the VFP's registers are used for float arguments. Your `libPDMFilter_CM4_GCC.a(pdm_filter.o)` library does this, but the rest of your program was compiled expecting floating point arguments to be passed in the CM4's registers. These two calling conventions are incompatible, as the code that was generated in say `main.o` expected that it could call functions in `pdm_filter.o` and pass float arguments in the CM4's registers. If this were to link without the error, then the output of this program would be completely arbitrary!

You might then follow up with:
> That's all well and good @theotherjimmy, but how do could I inspect what calling convention a given `.o` file is compiled for? How can I catch this sort of thing way before the link step?

That's part of the `arm-none-eabi-readelf` output, as described in this stack overflow question: https://stackoverflow.com/questions/20555594/how-can-i-know-if-an-arm-library-is-using-hardfp

And after running that command on your `main.o` you might ask:
> Why is `main.cpp` compiled to use CM4 registers for floating point arguments?

Looking in the tools that build your binary (specifically here: https://github.com/ARMmbed/mbed-os/blob/master/tools/toolchains/gcc.py#L74-L76), we (Mbed OS) configure the `softfp` floating point ABI.

The calling conventoins are discussed in the GCC Arm-specific documentation https://gcc.gnu.org/onlinedocs/gcc/ARM-Options.html in the section `-mfloat-abi=name`. Note that `softfp` uses hardware floating point. Also note that it is incompatible with `hard`.

From this point onward, I can't really help further without knowing the problem you're trying to solve. Could you explain what problem it is you're trying to solve, and what you have tried?

vrushalibhokare, 2018-11-15:
Thank you sir for help. I got the solution of this error . 
I solved this error using Mbed online compiler.

closed 2018-11-15 by vrushalibhokare