ARMmbed/mbed-cli #781: Custom target issue (STM32L451CEU6) fail to compile, HAL includes not recognized

Hoel, 2018-11-02:
Hello,
I already ported various custom targets with succes in the past (STM32F103CBT6, STM32F303CCT6, STM32L151C8T6, STM32L462CEU6), i now have to port our latest target STM32L451CEU6, unfortunately it fails completely, and depsite few hours of search i cannot get it to compile.

Here is what has been done:

- a target folder has been created (TARGET_L4680_L451), under targets/TARGET_STM/TARGET_STM32L451xE/
It contains the mandatory files, extracted from an STM32CubeMx project. The folder is attached.
[image]
[image]
[TARGET_STM32L451xE.zip](https://github.com/ARMmbed/mbed-cli/files/2541477/TARGET_STM32L451xE.zip)

- custom target has been added in targets.json

  

```
    "L4680_L451": {
        "inherits": ["FAMILY_STM32"],
        "supported_form_factors": ["ARDUINO"],
        "core": "Cortex-M4F",
        "extra_labels_add": ["STM32L4", "STM32L451xE", "STM32L451CE", "STM32L451xx"],
        "config": {
            "clock_source": {
                "help": "Mask value : USE_PLL_HSE_EXTC (need HW patch) | USE_PLL_HSE_XTAL (need HW patch) | USE_PLL_HSI | USE_PLL_MSI",
                "value": "USE_PLL_MSI",
                "macro_name": "CLOCK_SOURCE"
            },
            "lpticker_lptim": {
                "help": "This target supports LPTIM. Set value 1 to use LPTIM for LPTICKER, or 0 to use RTC wakeup timer",
                "value": 1
            }
        },
        "detect_code": ["7017"],
        "supported_toolchains": ["GCC_ARM"],
        "device_has_add": ["ANALOGOUT", "FLASH"],
        "release_versions": ["2", "5"],
        "device_name": "STM32L451CE",
        "bootloader_supported": true
    },
```
[targets.json.zip](https://github.com/ARMmbed/mbed-cli/files/2541509/targets.json.zip)

Target MCU has been added in mbed_rtx.h
[mbed_rtx.h.zip](https://github.com/ARMmbed/mbed-cli/files/2541513/mbed_rtx.h.zip)

- target has been added in tools/export/sw4stm32  init.py dictionnary (not relevant for compilation).

Unfortunately it refuses to compile past 0.4%, there is apparently something wrong with the includes system since the compiler doesnt see HAL peripherals (SPI, I2C, ADC, DAC) handles:

```
sudo mbed compile -m L4680_L451 -t GCC_ARM
Password:
[mbed] Auto-installing missing Python modules...
[Warning] @,: Compiler version mismatch: Have 4.8.4; expected version >= 6.0.0 and < 7.0.0
Building project mbed-os-example-blinky (L4680_L451, GCC_ARM)
Scan: mbed-os-example-blinky
Scan: env
Compile [  0.1%]: mbed_tz_context.c
Compile [  0.3%]: MCR20Drv.c
Compile [  0.4%]: AnalogIn.cpp
[Error] objects.h@58,5: 'RNG_HandleTypeDef' does not name a type
[Error] common_objects.h@53,5: 'SPI_HandleTypeDef' does not name a type
[Error] common_objects.h@92,5: 'I2C_HandleTypeDef' does not name a type
[Error] common_objects.h@120,5: 'ADC_HandleTypeDef' does not name a type
[Error] common_objects.h@131,5: 'DAC_HandleTypeDef' does not name a type
[ERROR] In file included from ./mbed-os/targets/TARGET_STM/TARGET_STM32L4/device.h:39:0,
                 from ./mbed-os/platform/platform.h:29,
                 from ./mbed-os/drivers/AnalogIn.h:19,
                 from ./mbed-os/drivers/AnalogIn.cpp:17:
./mbed-os/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L451xE/objects.h:58:5: error: 'RNG_HandleTypeDef' does not name a type
     RNG_HandleTypeDef handle;
     ^
In file included from ./mbed-os/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L451xE/objects.h:61:0,
                 from ./mbed-os/targets/TARGET_STM/TARGET_STM32L4/device.h:39,
                 from ./mbed-os/platform/platform.h:29,
                 from ./mbed-os/drivers/AnalogIn.h:19,
                 from ./mbed-os/drivers/AnalogIn.cpp:17:
./mbed-os/targets/TARGET_STM/TARGET_STM32L4/common_objects.h:53:5: error: 'SPI_HandleTypeDef' does not name a type
     SPI_HandleTypeDef handle;
     ^
./mbed-os/targets/TARGET_STM/TARGET_STM32L4/common_objects.h:92:5: error: 'I2C_HandleTypeDef' does not name a type
     I2C_HandleTypeDef handle;
     ^
./mbed-os/targets/TARGET_STM/TARGET_STM32L4/common_objects.h:120:5: error: 'ADC_HandleTypeDef' does not name a type
     ADC_HandleTypeDef handle;
     ^
In file included from ./mbed-os/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L451xE/objects.h:61:0,
                 from ./mbed-os/targets/TARGET_STM/TARGET_STM32L4/device.h:39,
                 from ./mbed-os/platform/platform.h:29,
                 from ./mbed-os/drivers/AnalogIn.h:19,
                 from ./mbed-os/drivers/AnalogIn.cpp:17:
./mbed-os/targets/TARGET_STM/TARGET_STM32L4/common_objects.h:131:5: error: 'DAC_HandleTypeDef' does not name a type
     DAC_HandleTypeDef handle;
     ^

[mbed] ERROR: "/usr/local/opt/python@2/bin/python2.7" returned error.
       Code: 1
       Path: "/Volumes/HD/XXXXX/XXXXXXX/FIRMWARE/L4680_L451CE_MBED_SRC/mbed-os-example-blinky"
       Command: "/usr/local/opt/python@2/bin/python2.7 -u /Volumes/HD/XXXXX/XXXXXXX/FIRMWARE/L4680_L451CE_MBED_SRC/mbed-os-example-blinky/mbed-os/tools/make.py -t GCC_ARM -m L4680_L451 --source . --build ./BUILD/L4680_L451/GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```
Is there is another dictionary somewhere i should add my target to get includes working? I have no idea what the problem is, the other custom targets have no problem.

If ST team could have a check and help to make this custom target work, it would be much appreaciated, at this point i have no idea what to try next.

Hoel, 2018-11-02:
I decided to add the HAL includes directly in device.h

```
#ifndef MBED_DEVICE_H
#define MBED_DEVICE_H

//=======================================
#define DEVICE_ID_LENGTH       24

#include "stm32l4xx_hal_rng.h"
#include "stm32l4xx_hal_dma.h"
#include "stm32l4xx_hal_adc.h"
#include "stm32l4xx_hal_dac.h"
#include "stm32l4xx_hal_spi.h"
#include "stm32l4xx_hal_i2c.h"

#include "objects.h"
/*  WORKAROUND waiting for mbed-os issue 4408 to be addressed */

#include "stm32l4xx_ll_usart.h"

#endif
```

It now compiles up to 83%

```
$ sudo mbed compile -m L4680_L451 -t GCC_ARM
[mbed] Auto-installing missing Python modules...
[Warning] @,: Compiler version mismatch: Have 4.8.4; expected version >= 6.0.0 and < 7.0.0
Building project mbed-os-example-blinky (L4680_L451, GCC_ARM)
Scan: mbed-os-example-blinky
Scan: env
Compile [  0.7%]: AnalogIn.cpp
....
Compile [ 83.0%]: PeripheralPins.c
[Error] PeripheralPins.c@51,54: 'GPIO_NOPULL' undeclared here (not in a function)
[Error] PinNamesTypes.h@98,33: invalid operands to binary & (have 'const struct PinMap *' and 'int')
```
then few thousand lines of errors are thrown, full log is include below
[built_log.txt](https://github.com/ARMmbed/mbed-cli/files/2541982/built_log.txt)

Hoel, 2018-11-02:
```
Compile [ 96.2%]: mbed_wait_api_rtos.cpp
[Warning] MbedCRC.h@36,32: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]
[Warning] mbed_wait_api_rtos.cpp@45,41: 'static osStatus rtos::Thread::wait(uint32_t)' is deprecated (declared at ./mbed-os/rtos/Thread.h:442): Static methods only affecting current thread cause confusion. Replaced by ThisThread::sleep_for. [since mbed-os-5.10] [-Wdeprecated-declarations]
Compile [ 96.3%]: stm32l4xx_hal_uart_ex.c
[Error] stm32l4xx_hal_uart_ex.c@490,27: 'USART_CR3_UCESM' undeclared (first use in this function)
[Error] stm32l4xx_hal_uart_ex.c@539,29: 'USART_CR3_UCESM' undeclared (first use in this function)
[ERROR] ./mbed-os/targets/TARGET_STM/TARGET_STM32L4/device/stm32l4xx_hal_uart_ex.c: In function 'HAL_UARTEx_EnableClockStopMode':
./mbed-os/targets/TARGET_STM/TARGET_STM32L4/device/stm32l4xx_hal_uart_ex.c:490:27: error: 'USART_CR3_UCESM' undeclared (first use in this function)
   huart->Instance->CR3 |= USART_CR3_UCESM;
                           ^
./mbed-os/targets/TARGET_STM/TARGET_STM32L4/device/stm32l4xx_hal_uart_ex.c:490:27: note: each undeclared identifier is reported only once for each function it appears in
./mbed-os/targets/TARGET_STM/TARGET_STM32L4/device/stm32l4xx_hal_uart_ex.c: In function 'HAL_UARTEx_DisableClockStopMode':
./mbed-os/targets/TARGET_STM/TARGET_STM32L4/device/stm32l4xx_hal_uart_ex.c:539:29: error: 'USART_CR3_UCESM' undeclared (first use in this function)
   huart->Instance->CR3 &= ~(USART_CR3_UCESM);
```

Hopefully the last error, related to USART_CR3_UCESM, however this is internal HAL files and by experience it is preferable not to mess with HAL files because when an error occurs here it is the sign of a more major issue. 
@bcostm ST team, please check my custom target folder and help with this issue.

JojoS62, 2018-11-02:
you are not using the HAL drinvers, they are not enabled in stm32l4xx.h:

```c
#if !defined  (USE_HAL_DRIVER)
/**
 * @brief Comment the line below if you will not use the peripherals drivers.
   In this case, these drivers will not be included and the application code will
   be based on direct access to peripherals registers
   */
  #define USE_HAL_DRIVER 
#endif /* USE_HAL_DRIVER */
```

then it will compile until system_clock.c, it looks like there are some modifications neccessary.

JojoS62, 2018-11-02:
remaining: 
- .\mbed-os\targets\TARGET_STM\TARGET_STM32L4\TARGET_STM32L451xE\device\us_ticker_data.h is missing
- PeripheralNames.h contains a double definition for UART4, try
```c
typedef enum {
    UART_1 = (int)USART1_BASE,
    UART_2 = (int)USART2_BASE,
    UART_3 = (int)USART3_BASE,
    UART_4 = (int)UART4_BASE,
    LPUART_1 = (int)LPUART1_BASE
} UARTName; 
```

Happy testing :)

JojoS62, 2018-11-02:
> So i should define USE_HAL_DRIVER?

I just compared to the other targets, all other have this #define uncommented in the stm32f4xx.h, like my code snippet.

Hoel, 2018-11-02:
Ohh, i didnt even realized the define in the conditional statement was actaully commented in my STM32L4xx.h 
that should explain why i had to includes all these HAL stuffs everywhere. Looks like i am close to successful compilation now. Thanks a lot for spotting this.

Hoel, 2018-11-02:
i removed my includes from device.h, now it compiles again but stall to the same point as before (USART_CR3_UCESM), i initally overlooked the second part of your answer, now trying to add the fix mentioned.

> ```
> /* MBED */
> #define USART_CR3_UCESM_Pos           (23U)
> #define USART_CR3_UCESM_Msk           (0x1U << USART_CR3_UCESM_Pos)            /*!< 0x00800000 */
> #define USART_CR3_UCESM               USART_CR3_UCESM_Msk                      /*!< Clock Enable in Stop mode */ 
> /* MBED */
> ```

```
Compile [ 95.3%]: stm32l4xx_hal_uart_ex.c
[Error] stm32l4xx_hal_uart_ex.c@490,27: 'USART_CR3_UCESM' undeclared (first use in this function)
[Error] stm32l4xx_hal_uart_ex.c@539,29: 'USART_CR3_UCESM' undeclared (first use in this function)
[ERROR] ./mbed-os/targets/TARGET_STM/TARGET_STM32L4/device/stm32l4xx_hal_uart_ex.c: In function 'HAL_UARTEx_EnableClockStopMode':
./mbed-os/targets/TARGET_STM/TARGET_STM32L4/device/stm32l4xx_hal_uart_ex.c:490:27: error: 'USART_CR3_UCESM' undeclared (first use in this function)
   huart->Instance->CR3 |= USART_CR3_UCESM;
                           ^
./mbed-os/targets/TARGET_STM/TARGET_STM32L4/device/stm32l4xx_hal_uart_ex.c:490:27: note: each undeclared identifier is reported only once for each function it appears in
./mbed-os/targets/TARGET_STM/TARGET_STM32L4/device/stm32l4xx_hal_uart_ex.c: In function 'HAL_UARTEx_DisableClockStopMode':
./mbed-os/targets/TARGET_STM/TARGET_STM32L4/device/stm32l4xx_hal_uart_ex.c:539:29: error: 'USART_CR3_UCESM' undeclared (first use in this function)
   huart->Instance->CR3 &= ~(USART_CR3_UCESM);
```

Hoel, 2018-11-02:
right, adding the register definition to STM32L4xx.h did the trick for UART_CR3_UCESM, now i have a last error on TIM5 in us_ticker_data.h since there is no TIM5 on this target (i scavanged the file from another nucleo target) hopefully it is just a matter of changing the TIM number
[image]

```
Compile [ 99.9%]: us_ticker.c
[Error] us_ticker_data.h@27,22: 'TIM5' undeclared (first use in this function)
[Warning] us_ticker.c@157,5: implicit declaration of function '__HAL_RCC_TIM5_CLK_ENABLE' [-Wimplicit-function-declaration]
[Warning] us_ticker.c@160,5: implicit declaration of function '__HAL_RCC_TIM5_FORCE_RESET' [-Wimplicit-function-declaration]
[Warning] us_ticker.c@161,5: implicit declaration of function '__HAL_RCC_TIM5_RELEASE_RESET' [-Wimplicit-function-declaration]
[Error] us_ticker_data.h@27,22: 'TIM5' undeclared (first use in this function)
[Error] us_ticker_data.h@28,22: 'TIM5_IRQn' undeclared (first use in this function)
[Error] us_ticker_data.h@27,22: 'TIM5' undeclared (first use in this function)
[Error] us_ticker_data.h@27,22: 'TIM5' undeclared (first use in this function)
[Error] us_ticker_data.h@27,22: 'TIM5' undeclared (first use in this function)
[Warning] us_ticker.c@215,1: control reaches end of non-void function [-Wreturn-type]
[ERROR] In file included from ./mbed-os/targets/TARGET_STM/us_ticker.c:19:0:
./mbed-os/targets/TARGET_STM/us_ticker.c: In function 'timer_irq_handler':
./mbed-os/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L451xE/device/us_ticker_data.h:27:22: error: 'TIM5' undeclared (first use in this function)
 #define TIM_MST      TIM5
```

Hoel, 2018-11-02:
After fixing us_ticker_dat, it finally compiles, now time to mess with the SW4STM32 project export / generation.

```
$ sudo mbed compile -m L4680_L451 -t GCC_ARM
Password:
[mbed] Auto-installing missing Python modules...
[Warning] @,: Compiler version mismatch: Have 4.8.4; expected version >= 6.0.0 and < 7.0.0
Building project mbed-os-example-blinky (L4680_L451, GCC_ARM)
Scan: mbed-os-example-blinky
Scan: env
Compile [100.0%]: us_ticker.c
Link: mbed-os-example-blinky
Elf2Bin: mbed-os-example-blinky
| Module           |         .text |       .data |        .bss |
|------------------|---------------|-------------|-------------|
| [fill]           |       53(+53) |       4(+4) |       7(+7) |
| [lib]/c.a        | 24524(+24524) | 2204(+2204) |     56(+56) |
| [lib]/gcc.a      |   3792(+3792) |       0(+0) |       0(+0) |
| [lib]/m.a        |       80(+80) |       0(+0) |       0(+0) |
| [lib]/misc       |     112(+112) |     16(+16) |     28(+28) |
| main.o           |       68(+68) |       4(+4) |     28(+28) |
| mbed-os/drivers  |     216(+216) |       4(+4) |   100(+100) |
| mbed-os/hal      |   1807(+1807) |       4(+4) |     68(+68) |
| mbed-os/platform |   3184(+3184) |   260(+260) |   197(+197) |
| mbed-os/rtos     |   3485(+3485) |   164(+164) | 4336(+4336) |
| mbed-os/targets  |   8115(+8115) |       4(+4) |   832(+832) |
| Subtotals        | 45436(+45436) | 2664(+2664) | 5652(+5652) |
Total Static RAM memory (data + bss): 8316(+8316) bytes
Total Flash memory (text + data): 48100(+48100) bytes

Image: ./BUILD/L4680_L451/GCC_ARM/mbed-os-example-blinky.bin
```

closed 2018-11-02 by Hoel