ARMmbed/mbed-cli #782: MBED python requirements issues ?

jeromecoutant, 2018-11-06:
Hi

Since https://github.com/ARMmbed/mbed-os/pull/8542,
python modules version check is more strict.
But I think this doesn't really work ?

1. missing module check

In my environment, urllib3 is installed.
But if I print pkg here:
https://github.com/ARMmbed/mbed-cli/blob/e99326300cc548b632132ba2e08da3fdc1c74e71/mbed/mbed.py#L1626

=> I always got urllib3[secure]

2. module version check

"pip install -r requirements.txt" command is executed only if "missing" is not empty:
https://github.com/ARMmbed/mbed-cli/blob/e99326300cc548b632132ba2e08da3fdc1c74e71/mbed/mbed.py#L1628

=> this means that if each package is present, module version is not checked
and https://github.com/ARMmbed/mbed-os/pull/8542 is not useful?

screamerbg, 2018-11-09:
Fix is available on mbed-cli master. Could you please try it out and let me know if that fixes the problem?

jeromecoutant, 2018-11-14:
Hi

1.8.3 release is not ok

### TEST

$ pip install colorama==0.3.9
$ mbed test xxx

OK

$ pip install colorama -U
Successfully installed colorama-0.4.0
$ mbed test xxx

==>  It crashes….

````
Traceback (most recent call last):
  File "C:\Python27\Scripts\mbedgt-script.py", line 6, in <module>
    from pkg_resources import load_entry_point
  File "c:\python27\lib\site-packages\pkg_resources\__init__.py", line 3123, in <module>
    @_call_aside
  File "c:\python27\lib\site-packages\pkg_resources\__init__.py", line 3107, in _call_aside
    f(*args, **kwargs)
  File "c:\python27\lib\site-packages\pkg_resources\__init__.py", line 3136, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "c:\python27\lib\site-packages\pkg_resources\__init__.py", line 580, in _build_master
    return cls._build_from_requirements(__requires__)
  File "c:\python27\lib\site-packages\pkg_resources\__init__.py", line 593, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "c:\python27\lib\site-packages\pkg_resources\__init__.py", line 781, in resolve
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'colorama<0.4,>=0.3' distribution was not found and is required by mbed-greentea
````

jeromecoutant, 2018-11-21:
Hi
Any update ?
As seen above, 1.8.3 release can crash.
Thx

bridadan, 2018-12-10:
Hi @jeromecoutant, sorry for the slow reply. I have a change up to bump the dependency version of colorama to include 0.4.x: https://github.com/ARMmbed/greentea/pull/291

jeromecoutant, 2018-12-10:
Hi
I don't have any specific issue with colorama...
I think there is some issue with the mbed script when version doesn't match with requirements.

bridadan, 2018-12-10:
> I don't have any specific issue with colorama...

I'm sure you don't 😄  but that was the version error you posted above.

> I think there is some issue with the mbed script when version doesn't match with requirements.

This isn't an issue with Mbed CLI in particular, this is just how pip handles dependencies. If another package (or a user) comes in and upgrades/downgrades an installed dependency, then the previously installed packages may break due to the mismatch in required version numbers.

The usual way to avoid this is to use [Python Virtual Environments](https://docs.python.org/3/tutorial/venv.html). That way the package has full control over dependency versions.

TacoGrandeTX, 2019-03-11:
@jeromecoutant Have you tried CLI v1.9.1 to see if https://github.com/ARMmbed/mbed-cli/pull/833 improves things for you?

closed 2019-03-14 by jeromecoutant