ARMmbed/mbed-cli #783: Error at importing project due to long PATH

MarceloSalazar, 2018-11-08:
Windows 10
Mbed CLI 1.8.2
Python 2.7.15

I'm unable to import an example app within a long path:

### Steps to reproduce:

```
C:\Users\ms\gfgsfgfgfsgf\sdgoingrgnr\gisknnsonjonjf\onjsonjsousoubgs\3>mbed import https://github.com/ARMmbed/simple-mbed-cloud-client-template-restricted -v
```

### Error output:

```

[mbed] Working path "C:\Users\ms\gfgsfgfgfsgf\sdgoingrgnr\gisknnsonjonjf\onjsonjsousoubgs\3" (directory)
[mbed] Importing program "simple-mbed-cloud-client-template-restricted" from "https://github.com/ARMmbed/simple-mbed-cloud-client-template-restricted" at latest revision in the current branch
[mbed] Trying to guess source control management tool. Supported SCMs: bld, git, hg
[mbed] Found matching cached repository in "C:\Users\ms\.mbed\mbed-cache\github.com\ARMmbed/simple-mbed-cloud-client-template-restricted"
[mbed] Carbon copy from "C:\Users\ms\.mbed\mbed-cache\github.com\ARMmbed/simple-mbed-cloud-client-template-restricted" to "C:\Users\ms\gfgsfgfgfsgf\sdgoingrgnr\gisknnsonjonjf\onjsonjsousoubgs\3\simple-mbed-cloud-client-template-restricted"
[mbed] Writing cache lock file C:\Users\ms\.mbed\mbed-cache\github.com\ARMmbed/simple-mbed-cloud-client-template-restricted\.lock\pid for pid 6304
[mbed] Cache lock file exists with my pid ("6304"). Cleaning up.
[mbed] Setting url to "https://github.com/ARMmbed/simple-mbed-cloud-client-template-restricted" in C:\Users\ms\gfgsfgfgfsgf\sdgoingrgnr\gisknnsonjonjf\onjsonjsousoubgs\3\simple-mbed-cloud-client-template-restricted
[mbed] Cleaning up Git index
[mbed] Update cached copy from remote repository
[mbed] Fetching revisions from remote repository to "simple-mbed-cloud-client-template-restricted"
[mbed] Exec "git fetch --all --tags" in "C:\Users\ms\gfgsfgfgfsgf\sdgoingrgnr\gisknnsonjonjf\onjsonjsousoubgs\3\simple-mbed-cloud-client-template-restricted"
Fetching origin
[mbed] Discarding local changes in "simple-mbed-cloud-client-template-restricted"
[mbed] Checkout "master" in simple-mbed-cloud-client-template-restricted
[mbed] Exec "git checkout master -f -q" in "C:\Users\ms\gfgsfgfgfsgf\sdgoingrgnr\gisknnsonjonjf\onjsonjsousoubgs\3\simple-mbed-cloud-client-template-restricted"
[mbed] Writing cache lock file C:\Users\ms\.mbed\mbed-cache\github.com\ARMmbed/simple-mbed-cloud-client-template-restricted\.lock\pid for pid 6304
[mbed] Cache lock file exists with my pid ("6304"). Cleaning up.
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #610e35ddc6d5
[mbed] Trying to guess source control management tool. Supported SCMs: bld, git, hg
[mbed] Found matching cached repository in "C:\Users\ms\.mbed\mbed-cache\github.com\ARMmbed/mbed-os"
[mbed] Carbon copy from "C:\Users\ms\.mbed\mbed-cache\github.com\ARMmbed/mbed-os" to "C:\Users\ms\gfgsfgfgfsgf\sdgoingrgnr\gisknnsonjonjf\onjsonjsousoubgs\3\simple-mbed-cloud-client-template-restricted\mbed-os"
[mbed] Writing cache lock file C:\Users\ms\.mbed\mbed-cache\github.com\ARMmbed/mbed-os\.lock\pid for pid 6304
[mbed] Cache lock file exists with my pid ("6304"). Cleaning up.
[mbed] Setting url to "https://github.com/ARMmbed/mbed-os" in C:\Users\ms\gfgsfgfgfsgf\sdgoingrgnr\gisknnsonjonjf\onjsonjsousoubgs\3\simple-mbed-cloud-client-template-restricted\mbed-os
[mbed] Cleaning up Git index
[mbed] Update cached copy from remote repository
[mbed] Fetching revisions from remote repository to "mbed-os"
[mbed] Exec "git fetch --all --tags" in "C:\Users\ms\gfgsfgfgfsgf\sdgoingrgnr\gisknnsonjonjf\onjsonjsousoubgs\3\simple-mbed-cloud-client-template-restricted\mbed-os"
Fetching origin
remote: Enumerating objects: 3111, done.
remote: Counting objects: 100% (3111/3111), done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 7984 (delta 3108), reused 3108 (delta 3108), pack-reused 4873
Receiving objects: 100% (7984/7984), 6.23 MiB | 8.22 MiB/s, done.
Resolving deltas: 100% (5416/5416), completed with 1034 local objects.
From https://github.com/ARMmbed/mbed-os
 * [new branch]          feature-hal-spec-usb-deiv -> origin/feature-hal-spec-usb-deiv
 + 05d77ab77...fddabf8ee feature-hal-spec-usb-device -> origin/feature-hal-spec-usb-device  (forced update)
 * [new branch]          feature-lorawan-1-1  -> origin/feature-lorawan-1-1
   f15dbf2c3..eea338c90  feature-posix        -> origin/feature-posix
 + b622d8df9...0f24dd183 feature-secure-time  -> origin/feature-secure-time  (forced update)
 + f973fb136...58cdab255 feature-watchdog     -> origin/feature-watchdog  (forced update)
 + 7765de7c9...26a3d121d feature_namespace    -> origin/feature_namespace  (forced update)
   8bf51be75..0fe636995  master               -> origin/master
   c53d51fe9..bf6f2c3c6  mbed-os-5.10         -> origin/mbed-os-5.10
   763f0f410..04790ce0b  release-candidate    -> origin/release-candidate
 t [tag update]          latest               -> latest
 * [new tag]             mbed-os-5.10.2       -> mbed-os-5.10.2
 * [new tag]             mbed-os-5.10.3       -> mbed-os-5.10.3
 * [new tag]             tools-release-5.10.2 -> tools-release-5.10.2
[mbed] Discarding local changes in "mbed-os"
[mbed] Discarding cached repository
[mbed] Exec "git clone https://github.com/ARMmbed/mbed-os C:\Users\ms\gfgsfgfgfsgf\sdgoingrgnr\gisknnsonjonjf\onjsonjsousoubgs\3\simple-mbed-cloud-client-template-restricted\mbed-os" in "C:\Users\ms\gfgsfgfgfsgf\sdgoingrgnr\gisknnsonjonjf\onjsonjsousoubgs\3\simple-mbed-cloud-client-template-restricted"
[... 22 lines cut ...]
[mbed] Exec "hg clone https://github.com/ARMmbed/mbed-os C:\Users\ms\gfgsfgfgfsgf\sdgoingrgnr\gisknnsonjonjf\onjsonjsousoubgs\3\simple-mbed-cloud-client-template-restricted\mbed-os" in "C:\Users\ms\gfgsfgfgfsgf\sdgoingrgnr\gisknnsonjonjf\onjsonjsousoubgs\3\simple-mbed-cloud-client-template-restricted"
abort: 'https://github.com/ARMmbed/mbed-os' does not appear to be an hg repository:
---%<--- (text/html; charset=utf-8)

<!DOCTYPE html>
<html lang="en">
  <head>
    <meta charset="utf-8">
  <link rel="dns-prefetch" href="https://assets-cdn.github.com">
  <link rel="dns-prefetch" href="https://avatars0.githubusercontent.com">
  <link rel="dns-prefetch" href="https://avatars1.githubusercontent.com">
  <link rel="dns-prefetch" href="https://avatars2.githubusercontent.com">
  <link rel="dns-prefetch" href="https://avatars3.githubusercontent.com">
  <link rel="dns-prefetch" href="https://github-cloud.s3.amazonaws.com">
  <link rel="dns-prefetch" href="https://user-images.githubusercontent.com/">

  <link crossorigin="anonymous" media="all" integrity="sha512-pO33jvcHJW+Is841g3EjvCxCYy5felC5kerQ4UrmoxEtKiPefXZ/Cqmo+guuBClWRTs0c0/rjDWAV5oELOxJtg==" rel="stylesheet" href="https://assets-cdn.github.com/assets/frameworks-595acde653bb1d1cb9b98b07b1be666b.css" />
  <link crossorigin="anonymous" media="all" integrity="sha512-USeSg1/V5yXpynnuqTWqJiQYCR3U9goMzYUFRYjUdphLYnXjlzTdVCcoL0LjFKE6OjKUx4GwJkDvuYhx0BXsUw==" rel="stylesheet" h
---%<---
!
[mbed] ERROR: Unable to clone repository (https://github.com/ARMmbed/mbed-os/#610e35ddc6d59f153173c1e7b2748cf96d6c9bcd)
---

C:\Users\ms\gfgsfgfgfsgf\sdgoingrgnr\gisknnsonjonjf\onjsonjsousoubgs\3>
```

screamerbg, 2018-11-08:
@MarceloSalazar I'm not sure how this is an Mbed CLI issue. Git clearly reports fatal errors due to filesystem limitations, e.g.
```
error: unable to create file features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_MCU_NRF51822/sdk/source/libraries/bootloader_dfu/hci_transport/hci_mem_pool_internal.h: Filename too long
error: unable to create file features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_MCU_NRF51822/sdk/source/softdevice/common/softdevice_handler/ant_stack_handler_types.h: Filename too long
error: unable to create file features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_MCU_NRF51822/sdk/source/softdevice/common/softdevice_handler/ble_stack_handler_types.h: Filename too long
error: unable to create file features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_MCU_NRF51822/sdk/source/softdevice/common/softdevice_handler/softdevice_handler.c: Filename too long
error: unable to create file features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_MCU_NRF51822/sdk/source/softdevice/common/softdevice_handler/softdevice_handler.h: Filename too long
error: unable to create file features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_MCU_NRF51822/sdk/source/softdevice/common/softdevice_handler/softdevice_handler_appsh.c: Filename too long
error: unable to create file features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_MCU_NRF51822/sdk/source/softdevice/common/softdevice_handler/softdevice_handler_appsh.h: Filename too long
error: unable to create file features/FEATURE_BOOTLOADER/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_K66F/TARGET_FRDM/mbed-bootloader-k66f-block_device-sotp-v3_4_0.bin: Filename too long
error: unable to create file features/FEATURE_BOOTLOADER/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_MCU_K64F/TARGET_FRDM/mbed-bootloader-k64f-block_device-sotp-v3_4_0.bin: Filename too long
error: unable to create file features/FEATURE_BOOTLOADER/targets/TARGET_STM/TARGET_STM32F2/TARGET_NUCLEO_F207ZG/mbed-bootloader-nucleo_f207zg-block_device-sotp-v3.4.0-3-gdbbb.bin: Filename too long
error: unable to create file features/FEATURE_BOOTLOADER/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F411xE/TARGET_NUCLEO_F411RE/mbed-bootloader-nucleo_f411re-block_device-sotp-v3_4_0.bin: Filename too long
error: unable to create file features/FEATURE_BOOTLOADER/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F429xI/TARGET_NUCLEO_F429ZI/mbed-bootloader-nucleo_f429zi-block_device-sotp-v3_4_0.bin: Filename too long
fatal: cannot create directory at 'features/FEATURE_BOOTLOADER/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_MODULE_UBLOX_ODIN_W2/TARGET_UBLOX_EVK_ODIN_W2': Filename too long
warning: Clone succeeded, but checkout failed.
You can inspect what was checked out with 'git status'
and retry the checkout with 'git checkout -f HEAD'
```

Therefore this is either:
* An issue to report against Mbed OS that the structure is far too deep, e.g. "flatten" Mbed OS a bit
* An issue against Microsoft that Windows has a limitation that is more and more apparent these days

I don't see anything that Mbed CLI can do to remedy the folder length problem.

MarceloSalazar, 2018-11-08:
Do you think we could possibly detect the path length and show an error accordingly?

cmonr, 2018-11-08:
@MarceloSalazar It's possible, but there are a few things I'm trying in parallel to see if we can get aroung the long base path.

This is the kind of error I was trying to reproduce as a part of IOTCORE-253 :)