ARMmbed/mbed-cli #784: `mbed new` creates a Git repository

SeppoTakalo, 2018-11-08:
Calling `mbed new .` creates an empty Git repository in the folder where I call it.

This is problematic if I'm calling it within Git repository already.

```
~/src/mbed-os-examples-docs_only$ git remote -v
origin	git@github.com:ARMmbed/mbed-os-examples-docs_only.git (fetch)
origin	git@github.com:ARMmbed/mbed-os-examples-docs_only.git (push)
~/src/mbed-os-examples-docs_only$ mkdir My_example
~/src/mbed-os-examples-docs_only$ cd My_example/
~/src/mbed-os-examples-docs_only/My_example$ mbed new .
[mbed] Creating new program "My_example" (git)
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at branch/tag "latest"
[mbed] Updating reference "mbed-os" -> "https://github.com/ARMmbed/mbed-os/#bf6f2c3c6434a6de9eb9511feffa5948b3d1f20f"
[mbed] Auto-installing missing Python modules...
~/src/mbed-os-examples-docs_only/My_example$ git remote -v
~/src/mbed-os-examples-docs_only/My_example$ git status
On branch master

No commits yet

Changes to be committed:
  (use "git rm --cached <file>..." to unstage)

	new file:   mbed-os.lib

Untracked files:
  (use "git add <file>..." to include in what will be committed)

	.mbed
	mbed_settings.py

```

bridadan, 2018-11-08:
@SeppoTakalo this is the expected behavior at the moment.

But you bring up an interesting point. We nest git repositories with mbed-cli all the time for **libraries**, but we don't really have expected behavior when placing a new program in an existing git repository.

One option I see at the moment is to provide a `--scm none` or `--no-scm` option for the `mbed new` command, would that work for you?

(as a workaround, just run `rm -rf .git` for now 😄)

bridadan, 2018-11-09:
@SeppoTakalo it's documented here: https://github.com/ARMmbed/mbed-cli#creating-a-new-program-for-mbed-os-5

Specifically this line:

> This creates a new folder "mbed-os-program", **initializes a new repository** and imports the latest revision of the mbed-os dependency to your program tree.

bridadan, 2018-11-09:
Woah, apparently you can already do `--scm none`! I just tried it locally to confirm no git repository was created.

bridadan, 2018-11-19:
@SeppoTakalo does the `--scm none` solve your issue?

SeppoTakalo, 2018-11-20:
Yes.

theotherjimmy, 2018-11-20:
Where would you expect to find this in the handbook @SeppoTakalo ?