ARMmbed/mbed-cli #785: mbed-cli defaults to hg for github URLs when they do not end with .git

bremoran, 2018-11-09:
mbed-cli appears so process .lib files in such a way that are assumed to be hg URLs unless there is a `.git` extension. For github URLs, this is unnecessary.

theotherjimmy, 2018-11-09:
I take it this is trivial to reproduce: just remove a `.git` and watch mbed-cli run hg?

bremoran, 2018-11-09:
It appears to be, but I only noticed because I didn't have hg installed.

Josaphat, 2018-11-12:
I've run into an issue that appears to be related.

When trying to follow the quick start guide, all of the following commands:
```
$ mbed import https://github.com/ARMmbed/mbed-os-example-blinky
$ mbed import git@github.com/ARMmbed/mbed-os-example-blinky
$ mbed import --protocol git git@github.com/ARMmbed/mbed-os-example-blinky
$ mbed import --protocol git git@github.com/ARMmbed/mbed-os-example-blinky.git
```

resulted in basically the same error:

```
[mbed] ERROR: Could not execute "hg clone --config progress.assume-tty=true https://github.com/ARMmbed/mbed-os-example-blinky /home/jos/Documents/mbed-experiment/mbed-os-example-blinky" in "/home/jos/Documents/mbed-experiment".
       You can verify that it's installed and accessible from your current path by executing "hg".
```

The problem disappeared when I installed mercurial. I'm on Arch Linux and installed mbed-cli using `pip3`.

The full output of one of the commands is here (all were essentially the same):

```
jos@ares ~/Documents/mbed-experiment % mbed import git@github.com/ARMmbed/mbed-os-example-blinky                                                                                                                                          :(
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting                                                                                                
---                                                                   
[mbed] Working path "/home/jos/Documents/mbed-experiment" (directory)                                                                                
[mbed] Program path "/home/jos/Documents/mbed-experiment"                                                                                                                                                                            
[mbed] Importing program "mbed-os-example-blinky" from "ssh://git@github.com/ARMmbed/mbed-os-example-blinky" at latest revision in the current branch                                                                                         
[mbed] ERROR: Could not execute "hg clone --config progress.assume-tty=true ssh://git@github.com/ARMmbed/mbed-os-example-blinky /home/jos/Documents/mbed-experiment/mbed-os-example-blinky" in "/home/jos/Documents/mbed-experiment".
       You can verify that it's installed and accessible from your current path by executing "hg".                                                                                                                                            
---                                                                                                                                                                                                                                           
[mbed] WARNING: Using Python 3 with Mbed OS 5.8 and earlier can cause errors with compiling, testing and exporting                                                                                                                            
---
```

kevinong0108, 2019-01-02:
https://github.com/ARMmbed/mbed-cli/pull/810
This patch can resolve this issue (maybe python 3.x only), but I have no time to fix push request check fail...

adbridge, 2019-11-01:
Apologies, this was closed in error by the automated bot - re-opening.

remicles, 2019-11-28:
I also got the `[mbed] ERROR: Could not execute "hg clone --config...` error when running `mbed new program` in a Python3 virtualenv, but not in a Python2 virtualenv.  I guess the warning should be heeded.
```
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting                                                                                                
```

amq, 2020-03-18:
Would be great if `mbed` detected `github.com` and defaulted to `git` in such case.