ARMmbed/mbed-cli #786: mbed-cli has broken python setuptools_scm_git_archive dependency

jahess, 2018-11-10:
Currently fails with python2 or python3...

Collecting pyOCD==0.12.0 (from mbed-host-tests<=1.5,>=1.1.2->-r requirements.txt (line 11))
  Using cached https://files.pythonhosted.org/packages/89/03/ba2b7c7573b384a92981a6a199abd88a688df7f2dbbee1eea872366b84c0/pyOCD-0.12.0.tar.gz
    Complete output from command python setup.py egg_info:
    Download error on https://pypi.python.org/simple/setuptools_scm_git_archive/: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1051) -- Some packages may not be found!
    Download error on https://pypi.python.org/simple/setuptools-scm-git-archive/: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1051) -- Some packages may not be found!
    Couldn't find index page for 'setuptools_scm_git_archive' (maybe misspelled?)
    Download error on https://pypi.python.org/simple/: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1051) -- Some packages may not be found!
    No local packages or working download links found for setuptools_scm_git_archive
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/_g/kx33hc0c8xjbyl0059bpf7_00000gn/T/pip-install-drhli1el/pyOCD/setup.py", line 78, in <module>
        include_package_data=True,  # include files from MANIFEST.in
      File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/setuptools/__init__.py", line 128, in setup
        _install_setup_requires(attrs)
      File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/setuptools/__init__.py", line 123, in _install_setup_requires
        dist.fetch_build_eggs(dist.setup_requires)
      File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/setuptools/dist.py", line 513, in fetch_build_eggs
        replace_conflicting=True,
      File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/pkg_resources/__init__.py", line 774, in resolve
        replace_conflicting=replace_conflicting
      File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/pkg_resources/__init__.py", line 1057, in best_match
        return self.obtain(req, installer)
      File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/pkg_resources/__init__.py", line 1069, in obtain
        return installer(requirement)
      File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/setuptools/dist.py", line 580, in fetch_build_egg
        return cmd.easy_install(req)
      File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/setuptools/command/easy_install.py", line 667, in easy_install
        raise DistutilsError(msg)
    distutils.errors.DistutilsError: Could not find suitable distribution for Requirement.parse('setuptools_scm_git_archive')
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /private/var/folders/_g/kx33hc0c8xjbyl0059bpf7_00000gn/T/pip-install-drhli1el/pyOCD/

jahess, 2018-11-10:
This is doubly sad as I have no intention of ever using OpenOCD.

bridadan, 2018-11-12:
@jahess can you post a bit about your environment? From the paths looks like you might be installing this on a Mac?

j3hill, 2018-11-13:
@jahess ... It looks as though you are running on a Mac, and we have seen some issues when systems are upgraded to Mojave. You may need to update your xcode developer tools by running "xcode-select --install" or "xcode-select --reset". Please also note that it is recommended to install the Mbed CLI (including Python) with the "--user" flag and without using "sudo". Here is a link to more information about manually installing the Mbed CLI on the Mac OS: https://os.mbed.com/docs/v5.10/tutorials/quick-start-offline.html#macos

j3hill, 2019-02-04:
@jahess - It it appears that you are able to build as referenced in this closed Github issue: https://github.com/ARMmbed/mbed-cli/issues/816

j3hill, 2019-02-04:
@bridadan , @theotherjimmy Please close this issue.

closed 2019-02-04 by bridadan