ARMmbed/mbed-cli #788: How to ignore error when compile ?

zeecy-zhao, 2018-11-14:
I cant find such build option that can ignore build error and compile continue.

theotherjimmy, 2018-11-14:
@zeecy-zhao There is no such option.

screamerbg, 2018-11-14:
@theotherjimmy Perhaps the fundamental question is whether there should be one. What do you think?

bridadan, 2018-11-14:
What would be the goal of ignoring build errors? If you continued to try and compile the project, what would happen when it got to the end? Linking the project not be possible.

@zeecy-zhao would you mind detailing your use case a bit more? If there was such an option, what would you expect the behavior to be?

zeecy-zhao, 2018-11-15:
@bridadan 
just like  "make -k"

>        -k, --keep-going
            Continue as much as possible after an error.  While the target that failed, and those that depend on it, cannot be remade, the other dependencies of these targets can be processed all the same.

kjbracey-arm, 2018-11-15:
I guess the use case is to plough on compiling as much as you can to keep the CPU occupied, while you fix the error you saw scrolling past, reducing the development cycle time as there's less remaining to compile after making the fix.

Would be annoying unless errors and progress indications were piped to different streams, or progress updates were reduced to just a bar or something. Or might work well with something like an Eclipse CDT error parser?

zeecy-zhao, 2018-11-15:
@kjbracey-arm
Even working with eclipse cdt, the parser result could be different with command.

bridadan, 2018-11-15:
Ah thanks for clarifying @zeecy-zhao. Would require some modifications to the toolchain logic in Mbed OS, but might not be too bad. Any thoughts @theotherjimmy?

theotherjimmy, 2018-11-15:
We could make it happen. I'm not planning on doing it, but I can help provide guidance to anyone that wants to contribute it.

closed 2019-10-31 by ciarmcom