ARMmbed/mbed-cli #791: Mbed CLI doesn't correctly handle target detection during `mbed dm update device -D <id> -m DETECT`

screamerbg, 2018-11-16:
See title

```
mbed dm update device -D 01671d2e968c0000000000010010038c -m DETECT -t ARM
[mbed] Auto-installing missing Python modules...
Traceback (most recent call last):
  File "D:\Work\smcc-example\mbed-os\tools\device_management.py", line 178, in <module>
    main()
  File "D:\Work\smcc-example\mbed-os\tools\device_management.py", line 173, in main
    }[options.action](options) or 0
  File "D:\Work\smcc-example\mbed-os\tools\device_management.py", line 79, in inner
    mcus = extract_mcus(MbedExtendedArgs(), options)
  File "D:\Work\smcc-example\mbed-os\tools\options.py", line 137, in extract_mcus
    args_error(parser, "argument -m/--mcu: {}".format(str(exc)))
  File "D:\Work\smcc-example\mbed-os\tools\utils.py", line 319, in args_error
    parser.error(message)
AttributeError: 'MbedExtendedArgs' object has no attribute 'error'
[mbed] ERROR: "c:\progra~1\python\python.exe" returned error.
       Code: 1
       Path: "D:\Work\smcc-example"
       Command: "c:\progra~1\python\python.exe -u D:\Work\smcc-example\mbed-os\tools\device_management.py update device -D 01671d2e968c0000000000010010038c --toolchain ARM --mcu DETECT --build .\BUILD\DETECT\ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
```

CC @theotherjimmy

theotherjimmy, 2018-11-16:
Yep. Good catch.

BlackstoneEngineering, 2019-03-04:
Any chance we could get this fixed? It'd make it a hellova lot easier to do tutorials if we had the `-m auto` command. Also can we make sure that it'll check the environment / global config if none is provided?

adbridge, 2019-11-01:
Apologies, this was closed in error by the automated bot - re-opening.

AshUK, 2020-05-13:
`mbed dm update device -D <device_id> -m <CUSTOM TARGET> -t ARM_GCC`

This command also fails when specifying a custom target