ARMmbed/mbed-cli #792: New command for coding style formatting

0xc0170, 2018-11-16:
`mbed format` command or similar for the formatting the code according to the Mbed OS coding style (using astyle , .astylerc option we provide as part of the OS, plus also .astyleignore file to ignore folders).

We have format code in the online IDE, it would match this new command (both behavior). 

Our next minor version 5.11 is going to have 0 coding style issues.

bulislaw, 2018-11-19:
+1 but format can be confusing, so I'd rather use astyle/style/code-format.

SeppoTakalo, 2018-11-21:
I would prefer `mbed astyle` or just a `.sh` script in the root that runs the Astyle.

I'm not fan of hiding behind `mbed ...` what we actually do.

mark-edgeworth, 2019-06-25:
@SeppoTakalo I agree. If formatting to some sort of mbed style is appropriate for contributors then this needs to be documented for contributors only and provided as a config file for whatever tool is doing the formatting. This is not going to be mbed CLI, so I'm closing this.

closed 2019-06-25 by mark-edgeworth