ARMmbed/mbed-cli #795: Auto-installing missing Python modules

janjongboom, 2018-11-28:
Every time when I compile, I get:

```
[mbed] Auto-installing missing Python modules...
```

Which takes multiple seconds. Running Mbed OS 5.10, Mbed CLI 1.8.2.

0xc0170, 2018-11-28:
Same here

Running master as of today (noticed this even in the last patch releases), Mbed CLI 1.8.2

Why are we getting this warning?

jeromecoutant, 2018-11-28:
Hi

See #782 
I think @janjongboom issue is the same

mbed CLI 1.8.3 is correcting this, but as tested in #782, crash can happen

0xc0170, 2018-12-03:
So this is a duplicate. @theotherjimmy / @bridadan Shall we close this?

jeromecoutant, 2018-12-04:
Proposition ?
- at first build, pip install requirements.txt is called with force update option
- requirements file is copied into BUILD directory (to compare files for next builds)
- pip is not called any more until the file is not changed

=> no more long "pip list" check, no more requirement parsing in script

Thx

JanneKiiskila, 2019-01-31:
Please fix this ASAP, it's a really annoying feature and I can only guess how much useless cycles it uses in our CIs.

screamerbg, 2019-01-31:
The python modules that are required for Mbed OS tools to run are changing over time, where more and various functionalities are added, e.g. icetea, mbed_cloud_sdk, manifest tool, but also python library modules.

To satisfy these changing requirements, it's part of the user flows (not CI flows) that Mbed CLI will check whether all required modules are installed and available for the user be able to run normally the Mbed OS tools - compile, test, export, etc. Therefore this is core part of the process and we cannot install requirements ONCE.

For your CI there's already a `--no-requirements` command switch. In the next hour I'll put up a PR that offers a configuration switch to disable the check and installation of requirements on config levels for users like yourself who love to have control of every single module in their space.

JanneKiiskila, 2019-02-01:
Those changes won't come in every day or every compile. If something breaks, we should check requirements then to fix it. The default behaviour should not be we check, but rather check when asked (in my opinion).

closed 2019-02-13 by ciarmcom