ARMmbed/mbed-cli #796: Flashing custom targets no longer working

t3k5g, 2018-11-29:
Hi, 

we are currently using the ST-Link V2.1 programmer part of a Nucleo board (see [here](https://jeelabs.org/book/1547a/) for reference) to programm our custom board. I recently updated mbed-cli and the automatic flashing with `mbed compile -f` is no longer working. The latest working version for me is 1.7.3. I think our problem is related to #717 which was introduced with mbed-cli v1.7.4. If I understand correctly #717 checks if the correct target is connected before flashing it and also chooses the correct target if multiple targets are connected. As our programmer is for a different board than the one we are compiling for mbed-cli gives the following error:

```
[mbed] ERROR: The target board you compiled for is not connected to your system.
       Please reconnect it and retry the last command.
```

Is there any way to deactivate  mbed-cli's behaviour to check if the right target is connected? If not would it be possible to add a command line argument which deactivates the check for custom targets?

bridadan, 2018-11-29:
Thanks for raising the issue @t3k5g. Your use case is an interesting one! As you can imagine, #717 solves issues for quite a few other folks that are using development boards with "known identities". While your case doesn't fall under this category, it is still valid.

You can override the identity of your board by using the mocking functionality in our tools: https://github.com/ARMmbed/mbed-ls#mocking-renaming-platforms

Give those instructions a try and let me know how it goes!

t3k5g, 2018-11-30:
Thanks a lot @bridadan for the quick reply. 

Using the mocking functionality does work for us and seems to be a good solution for our usecase. At first it didn't work because it didn't become clear to me from reading the readme that mocking does not work globally. After executing the command in my project folder, which creates a .mbedls-mock file, it did work however. So it seems this needs to be done for every project but that is ok for us.

Another solution I discovered is to use retargeting which is also described in the readme. Specifically one needs to retarget the platform_name . As this requires the full target_id which differs for every ST-Link programmer mocking seems to be the better solution though.

Thanks a lot again :)

closed 2018-11-30 by t3k5g

rvasquez6089, 2019-01-09:
I also am having the same issue. It would be nice if the script just flashed a binary to any "ST_PLACEHOLDER" without worrying if the target matches. Not sure what you have planned for the fix, but I look forward to it! For now I will try mocking as suggested, or reverting to an older and possibly "dumber" version of the mbed-cli. If i remember correctly, older versions would flash without checking if the target matches the connected device.