ARMmbed/mbed-cli #797: Compiling mbed-os as a library fails

aureliano2nd, 2018-11-29:
I am trying to follow the guidelines from the main readme of mbed-cli.

I want to share a single mbed-os library between various projects. I followed the guidelines and imported mbed-os. But when I try to compile mbed-os as a library (`mbed compile -t GCC_ARM -m NUCLEO_F446RE --library`), it fails with the following error :

    [mbed] Program path "c:\STM32Toolchain\projects\mbed-os"
    unhashable type: 'dict'

    Completed in: (0.00)s

    Build failures:
    * GCC_ARM::NUCLEO_F446RE

    [mbed] ERROR: "c:\python27\python.exe" returned error.
        Code: 1
        Path: "c:\STM32Toolchain\projects\mbed-os"
        Command: "c:\python27\python.exe -u c:\STM32Toolchain\projects\mbed-os\tools\build.py -t GCC_ARM -m NUCLEO_F446RE --source . --build .\BUILD\libraries\mbed-os\NUCLEO_F446RE\GCC_ARM"
        Tip: You could retry the last command with "-v" flag for verbose output
    ---

I also tried to compile from the projects directory supplying the --source option and it fails with the same error message.

Note : as you can spot from the file path, I am trying to get this working on windows (10).

Thanks in advance.

aureliano2nd, 2018-11-29:
I added the verbose output 

    c:\STM32Toolchain\projects>mbed compile -t GCC_ARM -m NUCLEO_F446RE --library --no-archive --source=mbed-os --build=..\mbed-os-build -v
    [mbed] Working path "c:\STM32Toolchain\projects" (program)
    [mbed] Exec "c:\python27\python.exe -u c:\STM32Toolchain\projects\mbed-os\tools\build.py -t GCC_ARM -m NUCLEO_F446RE --source mbed-os --build ..\mbed-os-build -v --no-archive" in "c:\STM32Toolchain\projects"
    Traceback (most recent call last):
    File "c:\STM32Toolchain\projects\mbed-os\tools\build.py", line 191, in <module>
        if Target.get_target(mcu).is_PSA_secure_target:
    File "c:\STM32Toolchain\projects\mbed-os\tools\targets\__init__.py", line 92, in wrapper
        if (func.__name__, args) not in CACHES:
    TypeError: unhashable type: 'dict'
    [mbed] ERROR: "c:\python27\python.exe" returned error.
        Code: 1
        Path: "c:\STM32Toolchain\projects"
        Command: "c:\python27\python.exe -u c:\STM32Toolchain\projects\mbed-os\tools\build.py -t GCC_ARM -m NUCLEO_F446RE --source mbed-os --build ..\mbed-os-build -v --no-archive"
        Tip: You could retry the last command with "-v" flag for verbose output
    ---

TacoGrandeTX, 2019-01-03:
@aureliano2nd Sorry for the delay.  Can you check that you are using a released version of Mbed?  We've successfully run that command on versions v5.9, v5.10 and v5.11 and the library is successfully built and stored in `\BUILD\libraries\sandbox\NUCLEO_F446RE\GCC_ARM` .  

Another thought - from the mbed-os folder you can run `pip install -r requirements.txt` to shore up any deficiencies in your environment.

If you're still having problems, what version of Mbed are you using?

linlingao, 2019-05-29:
We could not reproduce the issue. I'm closing this issue because requester has not responded with more info.

closed 2019-05-31 by ciarmcom