ARMmbed/mbed-cli #798: There seem to be an issue with "mbed new" command.  (OSX)

fivefishstudios, 2018-12-03:
Hi All! 

There seem to be an issue with "mbed new <projectname" 
The program compiles and seem to be uploaded to the device successfully, but it doesn't work.
I'm using the PwmOut sample program. (just blinking an LED)

I also noticed a lot of warnings generated during the build cycle. 

Now, the same program from the Online Compiler works just fine when the binary is downloaded to my device. 

So it seems the problem is with the /mbed-os directory. 

WORKAROUND:
I decided to delete the /mbed-os directory created initially by the "mbed new" command.
Then I re-added it again using 
mbed add mbed-os

I recompiled the program and noticed the previous warning messages are gone/reduced substantially. 

And when the program is flashed to my device, the program now works successfully! 

So it seems the /mbed-os directory initially created by the mbed new command is faulty?
And the only fix is to manually delete this /mbed-os directory and re-add it again using "mbed add mbed-os"

Can anyone else confirm this behavior?

Im running on OSX. My mbed-cli is version 1.8.3

fivefishstudios, 2018-12-03:
Seems my problem is I was trying to PwmOut the built-in LED on the DISC1 board (F4) that is not PWM capable. 

BUT, the output of the 2 compile operations are still different depending on whether the /mbed-os directory was created with the 'mbed new' command, or an 'mbed add mbed-os' command.   The former generates a lot of [Warning] messages, vs. the latter. 

Attached are 2 text files. 

[mbed-add-mbed-os-compiler-output.txt](https://github.com/ARMmbed/mbed-cli/files/2639821/mbed-add-mbed-os-compiler-output.txt)
[mbed-new-compiler-output.txt](https://github.com/ARMmbed/mbed-cli/files/2639822/mbed-new-compiler-output.txt)

The compile stats generated at the end are also different, Total Flash memory, list of modules, etc.

j3hill, 2019-03-11:
@fivefishstudios The MBED CLI offers --help information that explains the difference between: 

"mbed new" = "Creates a new mbed program if executed within a non-program location. Alternatively creates an mbed library if executed within an existing program. When creating new program, the latest mbed-os release will be downloaded/added unless --create-only is specified." 
"mbed add" = "Adds mbed library and its dependencies from a source control based URL (GitHub, Bitbucket, mbed.org) into an existing program."

They key point is that "mbed new" commands gets the latest version of the mbed OS, while the add gets the mbed library (+dependencies) specified by the URL in the command.

It appears that the difference between the 2 text files you sent is that they are different versions of the mbed-os. The "new" version appears to be newer than the "add" version. Another possibility is that they have different versions of the ".mbedignore" file which specifies what components/etc should not be compiled.

linlingao, 2019-06-13:
@j3hill appears to have answered your question. @fivefishstudios can you close it if you're satisfied with the answer?

closed 2019-10-31 by ciarmcom