ARMmbed/mbed-cli #801: mbed-os-example-wifi not working with DISCO_L475VG_IOT01A

ltancerel, 2018-12-10:
I could not get WIFI example working using CLI on DISCO_L475VG_IOT01A getting following UART output :

> WiFi example
> Mbed OS version 5.10.4 
> ERROR: No WiFiInterface found
This is because `WiFiInterface::get_default_instance()` returns null in the example.
After some search I downloaded the ISM43362 library , in _mbed-os_ directory:
```
cd mbed-os
git clone https://github.com/ARMmbed/wifi-ism43362.git

```

and the example did work.

I am quite rookie on mbed and I don't know what is the appropriate process to push this change, and I am not sure this is the appropriate place to put that library, I would rather put it into `mbed-os/features/FEATURE_BOOTLOADER/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L475xG/TARGET_DISCO_L475VG_IOT01A/` ?

linlingao, 2018-12-11:
Can you put the ISM43362 driver under mbed-os/components/wifi and submit a PR? Once you submit a PR, our experts will guide you through.

JanneKiiskila, 2019-01-31:
It actually does not matter in which folder the ISM43362 driver is, the DISCO example repository has it in the root folder of the example.

TacoGrandeTX, 2019-01-31:
@ltancerel As @JanneKiiskila has pointed out this should build fine out of the box, so perhaps there was some failure at the CLI you didn't notice.  We just repeated the README.md steps and were able to build okay.  Please review that after your import operation you have the correct folder structure shown below:
```
$ mbed import http://os.mbed.com/teams/ST/code/mbed-os-example-wifi/
[mbed] Importing program "mbed-os-example-wifi" from "https://os.mbed.com/teams/ST/code/mbed-os-example-wifi" at latest                               revision in the current branch
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #569159b784f7
[mbed] Adding library "wifi-ism43362" from "https://github.com/ARMmbed/wifi-ism43362" at rev #e4ecc27e87d9
[mbed] Adding library "wifi-x-nucleo-idw01m1" from "https://github.com/ARMmbed/wifi-x-nucleo-idw01m1" at rev #5871f7011d                              7f

rfulchie@abc MINGW64 /c/_cases/ghi_801
$ cd mbed-os-example-wifi/

rfulchie@abc MINGW64 /c/_cases/ghi_801/mbed-os-example-wifi
$ ls -l
total 71
-rw-r--r-- 1 rfulchie 1049089  3193 Jan 31 12:47 Jenkinsfile
-rw-r--r-- 1 rfulchie 1049089 11357 Jan 31 12:47 LICENSE
-rw-r--r-- 1 rfulchie 1049089  4750 Jan 31 12:47 main.cpp
-rw-r--r-- 1 rfulchie 1049089  2360 Jan 31 12:47 mbed_app.json
-rw-r--r-- 1 rfulchie 1049089   960 Jan 31 12:47 mbed_app_idw01m1.json
-rw-r--r-- 1 rfulchie 1049089   983 Jan 31 12:47 mbed_app_idw04a1.json
-rw-r--r-- 1 rfulchie 1049089  2544 Jan 31 12:47 mbed_app_ism43362.json
-rw-r--r-- 1 rfulchie 1049089  1374 Jan 31 12:53 mbed_settings.py
drwxr-xr-x 1 rfulchie 1049089     0 Jan 31 12:52 mbed-os/
-rw-r--r-- 1 rfulchie 1049089    77 Jan 31 12:47 mbed-os.lib
-rw-r--r-- 1 rfulchie 1049089  4377 Jan 31 12:47 README.md
drwxr-xr-x 1 rfulchie 1049089     0 Jan 31 12:52 wifi-ism43362/
-rw-r--r-- 1 rfulchie 1049089    84 Jan 31 12:47 wifi-ism43362.lib
drwxr-xr-x 1 rfulchie 1049089     0 Jan 31 12:52 wifi-x-nucleo-idw01m1/
-rw-r--r-- 1 rfulchie 1049089    90 Jan 31 12:47 wifi-x-nucleo-idw01m1.lib
```
The default mbed_app.json file can be used for the DISCO_L475VG_IOT01A.

linlingao, 2019-05-29:
We couldn't reproduce this issue. The requester has not provided more info. Will close this issue.

closed 2019-05-31 by ciarmcom