ARMmbed/mbed-cli #802: Add `-a <APIKEY` to `mbed dm update` sub command

BlackstoneEngineering, 2018-12-12:
## Problem
There is currently no way to programatically set the API key to use in the `mbed dm update` command. There is the `-a` option in the `mbed dm init` command, but not in the update command. This is both inconsistent and super annoying to use when trying to script with mbed-cli. 

## Soution
Add the `-a <APIKEY` sub command with the same format that is used on `mbed dm init -a`.

bridadan, 2019-03-04:
I just checked the following commands:

```
mbed dm init -d "herp.derp" --model-name "12345" -a <api key> -f -q
mbed dm update device -D <device id> -p BUILD\mbed-cloud-client-example_update.bin -a <api key>
```

Both seemed to execute fine (or at least, made it to the update step, I didn't actually have a device setup and ready). Can you confirm the issue?

BlackstoneEngineering, 2019-03-04:
@bridadan - has this been added since I filed this issue? Also did you make sure it wasnt pulling in the key from your environment?

bridadan, 2019-03-04:
> has this been added since I filed this issue?

I checked through the history of all the tools involved, but it looks like this has pretty much always been present (might be missing something).

> Also did you make sure it wasnt pulling in the key from your environment

Yup, it's not in my local or global config.

Could you try reproducing the issue? I was working with the latest Mbed CLI.

BlackstoneEngineering, 2019-03-04:
if it works them im happy.

closed 2019-03-04 by BlackstoneEngineering