ARMmbed/mbed-cli #804: export to cmake_gcc_arm generates include list in wrong order, doesn't add all libs to PRE_LINK command

bearsh, 2018-12-14:
I try to use to use the cmake_gcc_arm exporter with the BLE_Heartbeat example and encountered following problems:

- in mbed-os there are two files called `EventQueue.h`, one in `mbed-os/events/` the other in `mbed- os/features/FEATURE_BLE/ble/pal/`. looking at the generated include list, the latter mentioned path precedes the former resulting in a compile error as the `EventQueue.h` in the `FEATURE_BLE` is in another namespace and _ble_ specific. changing the order fixes the problem.

- the linking fails as the cryptocell related libraries are not specified in the `PRE_LINK` command. adding these libraries to `TARGET_LINK_LIBRARIES` fixes the problem. (the libs were added to `LD_SYS_LIBS` variable)

theotherjimmy, 2018-12-14:
@donatieng 

> in mbed-os there are two files called EventQueue.h, one in mbed-os/events/ the other in mbed- os/features/FEATURE_BLE/ble/pal/. 

This is a bug, probably in BLE. How did we miss the warning at the end of compile for so long?

-------------------------------

@bearsh Thanks for the bug report. Which version of Mbed OS and Mbed CLI are you using?

bearsh, 2018-12-17:
mbed: mbed-os-5.10.4, 2fd0c5cfb
mbed-cli: v1.8.3 from pip

evedon, 2019-05-22:
@donatieng Was the issue with a file called `EventQueue.h` in BLE fixed?

closed 2019-10-31 by ciarmcom

bearsh, 2019-10-31:
wow, one can resolve bugs or simply close them by adding a stupid comment (@ciarmcom), either way the stats will look great at the end. this really encourage people participating...
didn't @theotherjimmy clearly say 

> This is a bug, probably in BLE

(probably the issue itself has been solved when the ble stack switched to cordio, I don't no, but still the reason the bug was closed with is simply wrong)

adbridge, 2019-11-01:
@bearsh we had some issues with out automated bot yesterday which closed this. However as this is indeed thought to be a bug in BLE it should actually be an issue in the mbed-os repo not in the cli repo...If you are still seeing the issue with the latest version of mbed-os please re-raise the issue there. Many thanks.