ARMmbed/mbed-cli #805: mbed.h: No such file or directory with mbed new --mbedlib

matsujirushi, 2018-12-18:
`mbed new --mbedlib` not expand .h files.

Does `https://mbed.org/users/mbed_official/code/mbed/builds/3a7713b1edbc` has problem?

```
C:\mbed>mbed new test --mbedlib
[mbed] Creating new program "test" (git)
[mbed] Adding library "mbed" from "https://mbed.org/users/mbed_official/code/mbed/builds" at branch/tag "tip"
[mbed] Unpacking library build "3a7713b1edbc" in "C:\mbed\test\mbed"
[mbed] Unpacking library build "3a7713b1edbc" in "C:\mbed\test\mbed"
[mbed] Updating reference "mbed" -> "https://mbed.org/users/mbed_official/code/mbed/builds/3a7713b1edbc"
[mbed] Couldn't find build tools in your program. Downloading the mbed 2.0 SDK tools...

C:\mbed>cd test

C:\mbed\test>notepad main.cpp

C:\mbed\test>mbed compile -t GCC_ARM -m LPC11U35_401
Building project test (LPC11U35_401, GCC_ARM)
Scan: test
Compile [100.0%]: main.cpp
[Fatal Error] main.cpp@1,18: mbed.h: No such file or directory
[ERROR] .\main.cpp:1:18: fatal error: mbed.h: No such file or directory
 #include "mbed.h"
                  ^
compilation terminated.

[mbed] ERROR: "c:\python27\python.exe" returned error.
       Code: 1
       Path: "C:\mbed\test"
       Command: "c:\python27\python.exe -u C:\mbed\test\.temp\tools\make.py -t GCC_ARM -m LPC11U35_401 --source . --build .\BUILD\LPC11U35_401\GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
---

C:\mbed\test>
```

Timmmm, 2019-01-13:
I am also having this problem. In fact there is no `mbed.h` anywhere in the project tree. How is this supposed to work?

0xc0170, 2019-01-14:
> I am also having this problem. In fact there is no mbed.h anywhere in the project tree. How is this supposed to work?

@Timmmm Are you using the same command as reported above ? Just checking , we will review

Timmmm, 2019-01-14:
Yes:

```
PS C:\Users\Tim\Documents> mbed new blinky --mbedlib
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "C:\Users\Tim\Documents" (directory)
[mbed] Program path "C:\Users\Tim\Documents"
[mbed] Creating new program "blinky" (git)
[mbed] Adding library "mbed" from "https://mbed.org/users/mbed_official/code/mbed/builds" at branch/tag "tip"
[mbed] Downloading library build "3a7713b1edbc" (might take a while)
[mbed] Unpacking library build "3a7713b1edbc" in "C:\Users\Tim\Documents\blinky\mbed"
[mbed] Unpacking library build "3a7713b1edbc" in "C:\Users\Tim\Documents\blinky\mbed"
[mbed] Updating reference "mbed" -> "https://mbed.org/users/mbed_official/code/mbed/builds/3a7713b1edbc"
[mbed] Couldn't find build tools in your program. Downloading the mbed 2.0 SDK tools...

PS C:\Users\Tim\Documents> cd blinky

PS C:\Users\Tim\Documents\blinky> Add-Content 'main.cpp' @'
>> #include <mbed.h>
>> int main() {}
>> '@

PS C:\Users\Tim\Documents\blinky> mbed compile -t GCC_ARM -m ARCH_BLE
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "C:\Users\Tim\Documents\blinky" (program)
[ERROR] 'dict' object has no attribute 'has_key'
[mbed] ERROR: "c:\users\tim\appdata\local\programs\python\python37-32\python.exe" returned error.
       Code: 1
       Path: "C:\Users\Tim\Documents\blinky"
       Command: "c:\users\tim\appdata\local\programs\python\python37-32\python.exe -u C:\Users\Tim\Documents\blinky\.temp\tools\make.py -t GCC_ARM -m ARCH_BLE --source . --build .\BUILD\ARCH_BLE\GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
---
[mbed] WARNING: Using Python 3 with Mbed OS 5.8 and earlier can cause errors with compiling, testing and exporting
---
PS C:\Users\Tim\Documents\blinky> mbed compile -t GCC_ARM -m ARCH_BLE -v
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "C:\Users\Tim\Documents\blinky" (program)
[mbed] Exec "c:\users\tim\appdata\local\programs\python\python37-32\python.exe -u C:\Users\Tim\Documents\blinky\.temp\tools\make.py -t GCC_ARM -m ARCH_BLE --source . --build .\BUILD\ARCH_BLE\GCC_ARM -v" in "C:\Users\Tim\Documents\blinky"
Traceback (most recent call last):
  File "C:\Users\Tim\Documents\blinky\.temp\tools\make.py", line 293, in <module>
    ignore=options.ignore
  File "C:\Users\Tim\Documents\blinky\.temp\tools\build_api.py", line 510, in build_project
    profile = find_build_profile(path) or profile
  File "C:\Users\Tim\Documents\blinky\.temp\tools\build_profiles.py", line 37, in find_build_profile
    if MBED_SDK_REV_MAP.has_key(build):
AttributeError: 'dict' object has no attribute 'has_key'
[mbed] ERROR: "c:\users\tim\appdata\local\programs\python\python37-32\python.exe" returned error.
       Code: 1
       Path: "C:\Users\Tim\Documents\blinky"
       Command: "c:\users\tim\appdata\local\programs\python\python37-32\python.exe -u C:\Users\Tim\Documents\blinky\.temp\tools\make.py -t GCC_ARM -m ARCH_BLE --source . --build .\BUILD\ARCH_BLE\GCC_ARM -v"
       Tip: You could retry the last command with "-v" flag for verbose output
---
[mbed] WARNING: Using Python 3 with Mbed OS 5.8 and earlier can cause errors with compiling, testing and exporting
---
PS C:\Users\Tim\Documents\blinky> code .

# Fix school-boy Python 3 compatibility error.

PS C:\Users\Tim\Documents\blinky> mbed compile -t GCC_ARM -m ARCH_BLE -v
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "C:\Users\Tim\Documents\blinky" (program)
[mbed] Exec "c:\users\tim\appdata\local\programs\python\python37-32\python.exe -u C:\Users\Tim\Documents\blinky\.temp\tools\make.py -t GCC_ARM -m ARCH_BLE --source . --build .\BUILD\ARCH_BLE\GCC_ARM -v" in "C:\Users\Tim\Documents\blinky"
[Error] @,: Compiler version mismatch: Have 7.2.1; expected version >= 6.0.0 and < 7.0.0
Building project blinky (ARCH_BLE, GCC_ARM)
Scan: blinky
Macros: -DTARGET_CORTEX_M -DTARGET_NORDIC -DDEVICE_ANALOGIN=1 -DTARGET_NRF51822 -DTARGET_LIKE_MBED -DARM_MATH_CM0 -DDEVICE_PWMOUT=1 -DCMSIS_VECTAB_VIRTUAL -DDEVICE_I2C=1 -DDEVICE_PORTOUT=1 -DDEVICE_SPISLAVE=1 -DTARGET_MCU_NORDIC_16K -DTARGET_ARCH_BLE -DDEVICE_INTERRUPTIN=1 -DDEVICE_SERIAL=1 -DTOOLCHAIN_GCC -DDEVICE_PORTINOUT=1 -D__MBED__=1 -DTARGET_MCU_NRF51822 -DTARGET_MCU_NRF51_16K_S130 -DTARGET_M0 -D__CMSIS_RTOS -DTOOLCHAIN_GCC_ARM -D__CORTEX_M0 -DMBED_BUILD_TIMESTAMP=1547496645.3072412 -DTARGET_MCU_NRF51_16K_BASE -DNRF51 -DTARGET_RELEASE -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE="cmsis_nvic.h" -DTARGET_CORTEX -DDEVICE_PORTIN=1 -DTARGET_MCU_NRF51 -DTARGET_LIKE_CORTEX_M0 -DDEVICE_SLEEP=1 -DTARGET_MCU_NRF51_16K -DDEVICE_SPI=1 -DTARGET_FF_ARDUINO -D__MBED_CMSIS_RTOS_CM -DFEATURE_BLE=1
[... 10 lines cut ...]
    ignore=options.ignore
  File "C:\Users\Tim\Documents\blinky\.temp\tools\build_api.py", line 575, in build_project
    objects = toolchain.compile_sources(resources, sorted(resources.get_file_paths(FileType.INC_DIR)))
  File "C:\Users\Tim\Documents\blinky\.temp\tools\toolchains\__init__.py", line 413, in compile_sources
    return self.compile_seq(queue, objects)
[... cut ...]
       Command: "c:\users\tim\appdata\local\programs\python\python37-32\python.exe -u C:\Users\Tim\Documents\blinky\.temp\tools\make.py -t GCC_ARM -m ARCH_BLE --source . --build .\BUILD\ARCH_BLE\GCC_ARM -v"
       Tip: You could retry the last command with "-v" flag for verbose output
---
[mbed] WARNING: Using Python 3 with Mbed OS 5.8 and earlier can cause errors with compiling, testing and exporting
---
PS C:\Users\Tim\Documents\blinky> gci -recurse -filter "mbed.h"
PS C:\Users\Tim\Documents\blinky>
```

0xc0170, 2019-01-15:
@Timmmm where did you send the email to?

theotherjimmy, 2019-01-15:
@0xc0170 He sent the email to me.

screamerbg, 2019-02-01:
@theotherjimmy @Timmmm Any progress on this?

theotherjimmy, 2019-02-01:
Hi @screamerbg, Thanks for the reminder.

Hi @Timmmm, that file is part of the tools-release that we use for the online compiler. I have created a new tools release based on 5.11.3 that contains the diff you sent me in email. I'm going to test it as part of `mbed-sdk-tools` real quick, and push that into it's upstream when I'm done.

stefaandesmet2003, 2019-02-18:
any news on how to build for mbed2 from mbed-cli ? I'm trying to reproduce the online compiler output for a nrf51_dongle blinkie, and am stuck at this point. 
Thanks for your help!

toyowata, 2019-02-19:
@screamerbg @theotherjimmy 

When I download zipped archive from below:
https://os.mbed.com/users/mbed_official/code/mbed/archive/3a7713b1edbc.zip
I could not find following files in the archive:
```
drivers/*
hal/*
platform/*
mbed.h
targets.json
```
Also, same result (missing files in the mbed 2 library) by following command:
```
$ mbed import https://os.mbed.com/teams/mbed/code/mbed_blinky/
$ cd mbed_blinky
$ mbed deploy 
```

yennster, 2019-02-22:
@toyowata I replicated this issue as well:

```
~/Desktop/mbed-classic-program $ mbed deploy
[mbed] Working path "/Users/jenplu01/Desktop/mbed-classic-program" (program)
[mbed] Updating library "mbed" to rev #3a7713b1edbc
[mbed] Updating the mbed 2.0 SDK tools...
~/Desktop/mbed-classic-program $ mbed compile
[mbed] Working path "/Users/jenplu01/Desktop/mbed-classic-program" (program)
[Error] @,: Compiler version mismatch: Have 7.2.1; expected version >= 6.0.0 and < 7.0.0
Building project mbed-classic-program (K64F, GCC_ARM)
Scan: mbed-classic-program
Compile [100.0%]: main.cpp
[Fatal Error] main.cpp@1,10: mbed.h: No such file or directory
[ERROR] ./main.cpp:1:10: fatal error: mbed.h: No such file or directory
 #include "mbed.h"
          ^~~~~~~~
compilation terminated.

[mbed] ERROR: "/usr/local/opt/python@2/bin/python2.7" returned error.
       Code: 1
       Path: "/Users/jenplu01/Desktop/mbed-classic-program"
       Command: "/usr/local/opt/python@2/bin/python2.7 -u /Users/jenplu01/Desktop/mbed-classic-program/.temp/tools/make.py -t gcc_arm -m k64f --source . --build ./BUILD/K64F/GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

bridadan, 2019-02-22:
Hey folks, I took a look and indeed `mbed.h` is missing for the releases with the hashes `65be27845400` and `3a7713b1edbc`. I can confirm what @toyowata said (thanks very much!), the `e95d10626187` revision does contain the `mbed.h`. If you need to work offline, I would recommend using that revision if possible. If you need the latest versions, I have found that the online compiler still seems to be working. I will be pursuing this issue more, thanks for raising this (and sorry for the slow response).

CBazille, 2019-06-19:
Hey everyone, just dropping in on you all to say I've encountered this issue today with revision 65be27845400, trying to compile mbed OS 2 with mbed CLI for a STM32F042K6 eval board. Anything new on this issue ?

`mbed compile --clean --target nucleo_f042k6 --toolchain GCC_ARM --profile debug
[mbed] Working path "/home/corentin/dev/test-mbed2-1" (program)
WARNING: MBED_GCC_ARM_PATH set as environment variable but doesn't exist
[Error] @,: Compiler version mismatch: Have 7.3.1; expected version >= 6.0.0 and < 7.0.0
Building project test-mbed2-1 (NUCLEO_F042K6, GCC_ARM)
Scan: test-mbed2-1
Compile [100.0%]: main.cpp
[Fatal Error] main.cpp@1,10: mbed.h: No such file or directory
[ERROR] ./main.cpp:1:10: fatal error: mbed.h: No such file or directory
 #include "mbed.h"
          ^~~~~~~~
compilation terminated.

[mbed] ERROR: "/usr/bin/python" returned error.
       Code: 1
       Path: "/home/corentin/dev/test-mbed2-1"
       Command: "/usr/bin/python -u /home/corentin/dev/test-mbed2-1/.temp/tools/make.py -t GCC_ARM -m nucleo_f042k6 --profile debug --source . --build ./BUILD/NUCLEO_F042K6/GCC_ARM-DEBUG -c"
       Tip: You could retry the last command with "-v" flag for verbose output
---`

adbridge, 2019-11-01:
Apologies, this was closed in error by the automated bot - re-opening.

arjunak234, 2020-02-03:
How am i supposed to switch to the old revision? I changed my mbed.bld to 
>https://mbed.org/users/mbed_official/code/mbed/builds/e95d10626187

and ran `mbed deploy`

> [mbed] Working path "/home/arjun/src/030" (program)
>[mbed] Adding library "mbed" from "https://mbed.org/users/mbed_official/code/mbed/builds" at rev >#e95d10626187
>[mbed] Unpacking library build "e95d10626187" in "/home/arjun/src/030/mbed"
>[mbed] Updating the mbed 2.0 SDK tools...

but mbed still points to the latest revision

`ls mbed`
>65be27845400/

backinabit, 2020-02-17:
Yes, this is urgent. It is no longer possible to retrieve older builds.

The issue with the newer builds appears to be related to the zip command failing to complete the job. There are also a lot of missing targets in the archive.

N0rbert, 2020-03-06:
Still a problem.

This is issue should be fixed  ASAP to support Mbed OS 2 on small targets with 4 Kb of RAM.

lbdroid, 2020-03-14:
is there any way to bump up the priority of this problem? Its been OVER A YEAR and this entire mbed platform remains COMPLETELY USELESS AND BROKEN.

ajtag, 2020-03-14:
not sure about python2 but with python 3 on 18.04
export program from webui as gcc-arm
 edit line 39 in .temp/tools/build_profiles.py 
from 
    if MBED_SDK_REV_MAP.has_key(build):
to 
    if build in MBED_SDK_REV_MAP:

manually install pyelftools
I manually downloaded the latest build rev #65be27845400 as well unpacked that into /mbed/.bld

and i think it worked after that, sorry for a little vague, going off memory

arvoelke, 2020-04-06:
> I manually downloaded the latest build rev #65be27845400 as well unpacked that into /mbed/.bld

If you use the fix in #969 then the manual download should not be needed. With this fix, `mbed` will download the revision successfully; since you were using Python3, it was getting stuck in an infinite loop after the file was downloaded.

toyowata, 2020-04-07:
FYI.  The step below works for me.

```
# install pyenv
# install python 2.7.15
# install gcc-arm-none-eabi-6-2017-q2-update

pip install virtualenv
python -m virtualenv .venv2
.venv2\Script\activate

pip install mbed-cli==1.10.1 pyelftools jsonschema

mkdir test
cd test
hg clone https://os.mbed.com/users/mbed_official/code/mbed-sdk-tools/ tools
pip install -r .\tools\requirements.txt
git clone https://github.com/ARMmbed/mbed-dev

# create your main.cpp

mbed compile -m your_target -t gcc_arm
```

N0rbert, 2020-04-07:
It is all great, but `mbed.h` is still not found even with Python 2:

```
cd /tmp
$ mbed config -L
[mbed] Global config:
GCC_ARM_PATH=/home/user/Software/gcc-arm-none-eabi-9-2019-q4-major/bin

$ mbed new test-lib --mbedlib
[mbed] Working path "/tmp" (directory)
[mbed] Creating new program "test-lib" (git)
[mbed] Adding library "mbed" from "https://mbed.org/users/mbed_official/code/mbed/builds" at branch/tag "tip"
[mbed] Unpacking library build "65be27845400" in "/tmp/test-lib/mbed"
[mbed] Unpacking library build "65be27845400" in "/tmp/test-lib/mbed"
[mbed] Updating reference "mbed" -> "https://mbed.org/users/mbed_official/code/mbed/builds/65be27845400"
[mbed] Couldn't find build tools in your program. Downloading the mbed 2.0 SDK tools...

$ cd test-lib/
$ find . -name mbed.h
$ # (empty output)
```

```
cd /tmp
mbed import https://os.mbed.com/users/mbed_official/code/mbed_blinky/
cd mbed_blinky

$ mbed import https://os.mbed.com/users/mbed_official/code/mbed_blinky/
[mbed] Working path "/home/mate" (directory)
[mbed] Importing program "mbed_blinky" from "https://os.mbed.com/users/mbed_official/code/mbed_blinky" at latest revision in the current branch
[mbed] Adding library "mbed" from "https://mbed.org/users/mbed_official/code/mbed/builds" at rev #176b8275d35d
[mbed] Downloading library build "176b8275d35d" (might take a while)
[mbed] Unpacking library build "176b8275d35d" in "/home/mate/mbed_blinky/mbed"
[mbed] Couldn't find build tools in your program. Downloading the mbed 2.0 SDK tools...
[mbed] Auto-installing missing Python modules (jinja2, pyyaml, mbed_ls, mbed_host_tests, mbed_greentea, fuzzywuzzy)...
$ find . -name mbed.h
$ # (empty output)

```

So there is still nothing to compile. Any thoughts?

toyowata, 2020-04-07:
Hi @N0rbert 

### mbed new command

As reported [here](https://github.com/ARMmbed/mbed-cli/issues/805#issuecomment-466575993), library version `3a7713b1edbc` and `65be27845400` don't have an `mbed.h` file and other header files.

You can use the hash version `e95d10626187` or early one.
https://os.mbed.com/users/mbed_official/code/mbed/shortlog/

Open the `mbed.bld` by editor and change to...
```
http://mbed.org/users/mbed_official/code/mbed/builds/e95d10626187
```

Then deploy again with cache off.

```
$ mbed cache off
$ mbed deploy
```

### Importing the mbed_blinky

I think the `176b8275d35d` should be fine, but probably enabling library cache by Mbed-CLI seems working unexpectedly.  Maybe you can use the command below:

```
$ mbed cache off
$ mbed import https://os.mbed.com/users/mbed_official/code/mbed_blinky/
```

arvoelke, 2020-04-07:
Thanks @toyowata. Your suggestion to change `mbed.bld` to that line and then `mbed cache off` and `mbed deploy` fixed the issue for me. Confusingly, `ls mbed` will still use the old revision name. So don't be mislead by the name of the directory (this must be a separate bug?)! I also needed to revert my `CMSIS` repo to an older version (5.4.0 worked for what I was trying to compile).

N0rbert, 2020-04-07:
@toyowata thanks, but it fails.

```
$ cd /tmp 
$ hg clone https://os.mbed.com/users/mbed_official/code/mbed_blinky/
$ cd /tmp/mbed_blinky
$ mbed cache off
$ echo "http://mbed.org/users/mbed_official/code/mbed/builds/e95d10626187" > mbed.bld
$ cat mbed.bld 
http://mbed.org/users/mbed_official/code/mbed/builds/e95d10626187
$ mbed deploy
[mbed] Working path "/tmp/mbed_blinky" (program)
[mbed] Adding library "mbed" from "https://mbed.org/users/mbed_official/code/mbed/builds" at rev #e95d10626187
[mbed] Downloading library build "e95d10626187" (might take a while)
[mbed] Unpacking library build "e95d10626187" in "/tmp/mbed_blinky/mbed"
[mbed] Couldn't find build tools in your program. Downloading the mbed 2.0 SDK tools...
[mbed] Updating the mbed 2.0 SDK tools...         
```

ends with strange thing that *e95d10626187* is not a *e95d10626187*, it is *65be27845400*:

```
$ ls mbed
65be27845400
$ find . -name mbed.h
$ # (no output)
```

toyowata, 2020-04-08:
@arvoelke @N0rbert 

`mbed deploy` with `-v` option shows detail information.

```
$ mbed cache off
$ cat mbed.bld
http://mbed.org/users/mbed_official/code/mbed/builds/e95d10626187
$ mbed deploy -v
[mbed] Working path "/Users/toyowata/dev/mbed_blinky" (program)
[mbed] Adding library "mbed" from "https://mbed.org/users/mbed_official/code/mbed/builds" at rev #e95d10626187
[mbed] Trying to guess source control management tool. Supported SCMs: bld, hg, git
[mbed] Setting url to "http://mbed.org/users/mbed_official/code/mbed/builds/65be27845400" in /Users/toyowata/dev/mbed_blinky/mbed
[mbed] Downloading library build "e95d10626187" (might take a while)
^C[mbed] User aborted!
$ mbed ls
[mbed] Working path "/Users/toyowata/dev/mbed_blinky" (program)
mbed_blinky (#3451e643c64c, tag: tip)
`- mbed (#65be27845400)
```

hmm, it seems the Mbed-CLI download latest release of the library.

Now, my suggestion to use this step:
https://github.com/ARMmbed/mbed-cli/issues/805#issuecomment-610120652