ARMmbed/mbed-cli #806: Minimum setuptools version needed

cmonr, 2018-12-18:
If a user installs Python 2.7.11 (the minimum install req according to the docs), setuptools is incredibly outdated. Solution was to update setuptools with `pip install -U setuptools`.

Stdout of issue: 
```
c:\python27\lib\distutils\dist.py:267: UserWarning: Unknown distribution option: 'python_requires'
  warnings.warn(msg)
c:\python27\lib\distutils\dist.py:267: UserWarning: Unknown distribution option:  'long_description_content_type'
   warnings.warn(msg)
```

CC @pan-, who found the problem with a fresh version of 2.7.11 on Windows.

##### Note

Not sure if issue should be opened here or in PyOCD, which is what uses those two keys.

bridadan, 2019-01-02:
I'm not aware of a way to specify a minimum setuptools version in `setup.py`, are you @cmonr? A quick search didn't turn up anything either. We could just document something along the lines of:

> Ensure you have the latest Python setuptools: `pip install --upgrade setuptools`
> Install mbed-cli: `pip install mbed-cli`

screamerbg, 2019-02-01:
@cmonr Are you happy to issue a PR?

closed 2019-10-31 by ciarmcom