ARMmbed/mbed-cli #807: Require a way to build Mbed OS 2 applications using a custom built mbed library (for a custom platform)

deece, 2018-12-20:
I am currently trying to port some Eclipse projects that use a manually built mbed library over to mbed-cli.

I need a custom library as I have custom device options & linker scripts (one is for a bootloader, so I need a tight binary, with a number of features disabled & LTO enabled to get the size down, the other has a custom linker script to place the binary at the offset that the bootloader expects).

When creating a new mbed project with --mbedlib, there seems to be no way to avoid using a precompiled library and instead either build one with my custom options, or provide a path to my own pre-built library.

deece, 2018-12-20:
Update: I've bodged a solution by adding "5" to the releases for my platform, then using .mbedignore to disable components that don't compile. It's pretty hacky...

naveenkaje, 2018-12-20:
Hi @deece 
Thanks for bringing this to our attention. You are right. That's the current behavior and we are aware of it. We hope to address these pain points in the next major release.

closed 2019-12-04 by mark-edgeworth