ARMmbed/mbed-cli #809: While running testcases using icetea getting below error

Vanireddy6, 2018-12-26:
10:10:50.220 | TC     MainThread: setUpBench Throw exception
Traceback (most recent call last):
  File "c:\python27\lib\site-packages\icetea_lib\bench.py", line 1119, in run
    self.__setup_bench()
  File "c:\python27\lib\site-packages\icetea_lib\bench.py", line 1360, in __setup_bench
    self.__init_duts()
  File "c:\python27\lib\site-packages\icetea_lib\bench.py", line 823, in __init_duts
    raise DutConnectionError("Dut cli failed to initialize within set timeout!")
DutConnectionError: Dut cli failed to initialize within set timeout!

0xc0170, 2019-01-02:
Someone will review this soon. I do not see what is being  tested and how - basically, how to reproduce this error.

0xc0170, 2019-01-02:
Closing as insufficient issue report, please add more details and reopen

closed 2019-01-02 by 0xc0170