ARMmbed/mbed-cli #811: compile always fails with "RefResolutionError: addinfourl instance has no attribute '__exit__'"

bikeNomad, 2018-12-31:
I'm using version 1.8.3 of the mbed-cli from Github. I have Python 2.7.10 installed on Mac OS 10.13.6.
I followed the mbed-cli installation directions, and then imported one of my new programs from mbed.org and tried to compile it.
The first time, I got an error due to missing the `jsonschema` package. I then installed that.

Unfortunately, every time I try to compile I get the same Python error (##RefResolutionError: addinfourl instance has no attribute '__exit__'##).

Could someone please tell me how I could fix this?
I've also posted this to the Questions on the mbed web site because I wasn't sure where to go for an answer.

Here's the verbose output:
```
[ned] /Users/ned/src/stepperMotor/StepperMotorTest % mbed compile -vv
[mbed-99820] Working path "/Users/ned/src/stepperMotor/StepperMotorTest" (program)
[mbed-99820] Exec "/usr/local/opt/python@2/bin/python2.7 -m pip list -l" in "/Users/ned/src/stepperMotor/StepperMotorTest"
Package                       Version
----------------------------- -------------
affine                        2.2.2
appdirs                       1.4.3
attrs                         18.2.0
backports.functools-lru-cache 1.5
beautifulsoup4                4.7.0
certifi                       2018.11.29
chardet                       3.0.4
colorama                      0.4.1
dblatex                       0.3.4
Django                        1.10.5
enum34                        1.1.6
fasteners                     0.14.1
freetype-py                   1.0.2
funcsigs                      1.0.2
functools32                   3.2.3.post2
future                        0.16.0
fuzzywuzzy                    0.17.0
GDAL                          2.3.2
gerbmerge                     1.8
gevent                        1.0.1
greenlet                      0.4.5
hidapi                        0.7.99.post21
idna                          2.6
image                         1.5.5
intelhex                      2.2.1
intervaltree                  2.1.0
Jinja2                        2.10
jsonschema                    3.0.0a3
junit-xml                     1.8
kipart                        0.1.33
libusb1                       1.3.0
lockfile                      0.12.2
MarkupSafe                    1.0
mbed-cli                      1.8.3
mbed-greentea                 1.5.0
mbed-host-tests               1.4.4
mbed-ls                       1.4.6
mbed-tools                    0.1.14
mercurial                     4.8
mock                          2.0.0
monotonic                     1.5
nose                          1.3.6
numpy                         1.11.3
olefile                       0.44
pbr                           4.0.4
Pillow                        4.0.0
pip                           18.1
pip-upgrade-outdated          1.2
prettytable                   0.7.2
pycairo                       1.18.0
pycollada                     0.4.1
pycups                        1.9.68
pyelftools                    0.25
pyglet                        1.2.2
pyocd                         0.14.3
pyparsing                     2.0.3
pyqtgraph                     0.10.0
pyrsistent                    0.14.8
pyrtlsdr                      0.2.0
pyserial                      3.4
python-dateutil               1.5
python-dbusx                  0.9
python-poppler-qt4            0.18.1
python-tdbus                  0.9
python-upconvert              0.8.9
python-usbtmc                 0.6
pytz                          2015.2
pyusb                         1.0.2
PyYAML                        3.12
requests                      2.18.4
setuptools                    39.2.0
SimpleParse                   2.1.1a2
six                           1.12.0
sortedcontainers              2.0.4
soupsieve                     1.6
UNKNOWN                       0.0.0
urllib3                       1.22
websocket-client              0.48.0
wheel                         0.31.1
[mbed-99820] Exec "/usr/local/opt/python@2/bin/python2.7 -u /Users/ned/src/stepperMotor/StepperMotorTest/.temp/tools/make.py -t GCC_ARM -m KL25Z --source . --build ./BUILD/KL25Z/GCC_ARM -v" in "/Users/ned/src/stepperMotor/StepperMotorTest"
Building project StepperMotorTest (KL25Z, GCC_ARM)
Scan: StepperMotorTest
Traceback (most recent call last):
  File "/Users/ned/src/stepperMotor/StepperMotorTest/.temp/tools/make.py", line 293, in <module>
    ignore=options.ignore
  File "/Users/ned/src/stepperMotor/StepperMotorTest/.temp/tools/build_api.py", line 568, in build_project
    src_paths, toolchain, inc_dirs=inc_dirs)
  File "/Users/ned/src/stepperMotor/StepperMotorTest/.temp/tools/resources/__init__.py", line 518, in scan_with_toolchain
    toolchain.config.load_resources(self)
  File "/Users/ned/src/stepperMotor/StepperMotorTest/.temp/tools/config/__init__.py", line 1085, in load_resources
    f.path for f in resources.get_file_refs(FileType.JSON)
  File "/Users/ned/src/stepperMotor/StepperMotorTest/.temp/tools/config/__init__.py", line 539, in add_config_files
    errors = sorted(validator.iter_errors(cfg))
  File "/usr/local/lib/python2.7/site-packages/jsonschema-3.0.0a3-py2.7.egg/jsonschema/validators.py", line 280, in iter_errors
    for error in errors:
  File "/usr/local/lib/python2.7/site-packages/jsonschema-3.0.0a3-py2.7.egg/jsonschema/_validators.py", line 400, in properties
    schema_path=property,
  File "/usr/local/lib/python2.7/site-packages/jsonschema-3.0.0a3-py2.7.egg/jsonschema/validators.py", line 296, in descend
    for error in self.iter_errors(instance, schema):
  File "/usr/local/lib/python2.7/site-packages/jsonschema-3.0.0a3-py2.7.egg/jsonschema/validators.py", line 280, in iter_errors
    for error in errors:
  File "/usr/local/lib/python2.7/site-packages/jsonschema-3.0.0a3-py2.7.egg/jsonschema/_validators.py", line 310, in ref
    scope, resolved = validator.resolver.resolve(ref)
  File "/usr/local/lib/python2.7/site-packages/jsonschema-3.0.0a3-py2.7.egg/jsonschema/validators.py", line 700, in resolve
    return url, self._remote_cache(url)
  File "build/bdist.macosx-10.12-x86_64/egg/functools32/functools32.py", line 400, in wrapper
    result = user_function(*args, **kwds)
  File "/usr/local/lib/python2.7/site-packages/jsonschema-3.0.0a3-py2.7.egg/jsonschema/validators.py", line 710, in resolve_from_url
    raise RefResolutionError(exc)
RefResolutionError: addinfourl instance has no attribute '__exit__'
[mbed-99820] ERROR: "/usr/local/opt/python@2/bin/python2.7" returned error.
       Code: 1
       Path: "/Users/ned/src/stepperMotor/StepperMotorTest"
       Command: "/usr/local/opt/python@2/bin/python2.7 -u /Users/ned/src/stepperMotor/StepperMotorTest/.temp/tools/make.py -t GCC_ARM -m KL25Z --source . --build ./BUILD/KL25Z/GCC_ARM -v"
```

0xc0170, 2019-01-02:
This looks like json error ? Is it this bug that was fixed but not yet released: https://github.com/Julian/jsonschema/issues/478 ?

bridadan, 2019-01-02:
It's also worth noting our documented minimum Python version supported is Python 2.7.11.

bikeNomad, 2019-01-02:
> This looks like json error ? Is it this bug that was fixed but not yet released: [Julian/jsonschema#478](https://github.com/Julian/jsonschema/issues/478) ?

I looked in the call stack and didn't find a `with` clause, but I could have missed it.

bikeNomad, 2019-01-02:
> It's also worth noting our documented minimum Python version supported is Python 2.7.11.

I will try updating my Python version.

bridadan, 2019-01-11:
Hi @bikeNomad any luck?

evedon, 2019-05-24:
Closing due to inactivity. Please re-open if you need additional assistance.

closed 2019-05-24 by evedon