ARMmbed/mbed-cli #812: Create CLI argument to define arbitrary working dir

AlexFBP, 2019-01-01:
I know that the idea of `mbed compile` is to be in the same working folder, *like Git*; ok, this looks good, but for Travis CI buildings, this forces to add unnecesary "scripts" -Note the quotes (...)-

Why not to...

1. add an argument to the CLI of `mbed compile` like `--workdir=WORKDIR`, to:
2. `cd $WORKDIR`
3. current mbed behavior, and finally
4. `cd ..`?

This behavior would be the same like `make` does with `make -C`, or the same that `platformio run` has with `platformio run -d`; thus allowing Travis to also switch based on `env:` items.

AlexFBP, 2019-01-01:
As another alternative, `pushd $WORKDIR` and `popd` can be used instead of `cd $WORKDIR` and `cd ..`

bridadan, 2019-01-02:
Seems reasonable to me, thoughts @theotherjimmy?

screamerbg, 2019-02-01:
@theotherjimmy Agreed with Jimmy.

@AlexFBP Have you tried running `mbed new --program --create-only` in the folder you'd like to use as root, before running the `mbed compile` command?

AlexFBP, 2019-02-02:
I guess it would the same focus @screamerbg , this adds an additional step to CI :thinking: (sorry if that did sound lazy :sweat_smile: ), but I'll try that.

The problem lies when a GitHub repo [like this](https://github.com/AsharaStudios/GuruxDLMS.c/tree/master/mbed) is meant to be a library, holding many examples, that are currently folders with mbed projects inside, which of course you want to know if all of them are working. As example, [please check this .travis.yml](https://github.com/AsharaStudios/GuruxDLMS.c/blob/d08982b58514fe2b2fc7f5148e58b34bc382d983/.travis.yml)

This additional param would help also to simplify the build matrix

closed 2019-10-31 by ciarmcom