ARMmbed/mbed-cli #814: While running testcases using icetea getting below error

Vanireddy6, 2019-01-03:
Traceback (most recent call last):
  File "c:\python27\lib\site-packages\icetea_lib\bench.py", line 1119, in run
    self.__setup_bench()
  File "c:\python27\lib\site-packages\icetea_lib\bench.py", line 1360, in __setup_bench
    self.__init_duts()
  File "c:\python27\lib\site-packages\icetea_lib\bench.py", line 823, in __init_duts
    raise DutConnectionError("Dut cli failed to initialize within set timeout!")
DutConnectionError: Dut cli failed to initialize within set timeout!

**Reproducible Steps**:
1.Import ble_beacon example
2.And run Icetea test using below command:
>mbed test -m DISCO_L475VG_IOT01A -t GCC_ARM --icetea -n test_cmdline -DICETEA_EXAMPLE_ENABLED

OPpuolitaival, 2019-01-09:
I did try this in two ways

**First:**
`git clone git@github.com:ARMmbed/mbed-os.git`
`cd mbed-os`
`mbed test -m DISCO_L475VG_IOT01A -t GCC_ARM --icetea -n test_cmdline -DICETEA_EXAMPLE_ENABLED`
=> This seems to work

**Second:**
`mbed new .`
`mbed deploy BLE_Beacon`
`mbed test -m DISCO_L475VG_IOT01A -t GCC_ARM --icetea -n test_cmdline -DICETEA_EXAMPLE_ENABLED`
=> [mbed] WARNING: Cannot run icetea tests. Current folder does not contain TEST_APPS folder.

Which version of mbed-cli you have?

screamerbg, 2019-02-01:
@Vanireddy6 Are you able to follow @OPpuolitaival's instructions?

OPpuolitaival, 2019-10-10:
No feedback. Closing

closed 2019-10-10 by OPpuolitaival