ARMmbed/mbed-cli #815: export -i gnuarmeclipse build fails

jahess, 2019-01-03:
Exporting mbed-os-example-mesh-minimal with -i gnuarmeclipse and -m KW24D results in a project that will not link.  The error from eclipse is:

/Users/jahess/opt/gcc-arm-none-eabi-7-2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: mbed-os-example-mesh-minimal.elf section `.text' will not fit in region `m_text'
/Users/jahess/opt/gcc-arm-none-eabi-7-2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: region m_text overflowed with text and data
/Users/jahess/opt/gcc-arm-none-eabi-7-2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: region `m_text' overflowed by 228812 bytes
collect2: error: ld returned 1 exit status
make: *** [mbed-os-example-mesh-minimal.elf] Error 1

The mbed_app.json was modified to work with the KW24D as shown below.  This same setup works with "mbed compile."  Obviously there are a compiler/linker option getting dropped.  The include directory list for individual files is also extremely long.

For the successful mbed compile the following global options were in affect:
GNU_ARM_PATH=/Users/jahess/opt/gcc-arm-none-eabi-6-2017-q2-update/bin
TOOLCHAIN=GCC_ARM

diff --git a/mbed_app.json b/mbed_app.json
index 8de7119..df32d68 100644
--- a/mbed_app.json
+++ b/mbed_app.json
@@ -1,5 +1,9 @@
 {
     "config": {
+        "storage-device":{
+            "help": "Values for storage (Thread) devices are: MESH_NVM_HEAP, MESH_NVM_SD_CARD and MESH_NVM_NONE",
+            "value": "MESH_NVM_NONE"
+        },
         "enable-led-control-example": true,
         "LED": "NC",
         "BUTTON": "NC",
@@ -7,17 +11,15 @@
     },
     "target_overrides": {
         "*": {
-            "nanostack.configuration": "lowpan_router",
-            "nsapi.default-mesh-type": "LOWPAN",
-            "mbed-mesh-api.6lowpan-nd-panid-filter": "0xffff",
-            "mbed-mesh-api.6lowpan-nd-channel-page": 0,
-            "mbed-mesh-api.6lowpan-nd-channel": 12,
-            "mbed-mesh-api.6lowpan-nd-channel-mask": "(1<<12)",
-            "mbed-mesh-api.heap-size": 14000,
+            "nanostack.configuration": "thread_router",
+            "nsapi.default-mesh-type": "THREAD",
             "mbed-trace.enable": false,
+            "mbed-mesh-api.heap-size": 30000,
+            "mbed-mesh-api.thread-device-type": "MESH_DEVICE_TYPE_THREAD_ROUTER",
             "platform.stdio-convert-newlines": true,
             "platform.stdio-baud-rate": 115200,
-            "atmel-rf.provide-default": true,
+            "mbed-mesh-api.thread-use-static-link-config": true,
+            "atmel-rf.provide-default": false,
             "mcr20a.provide-default": false,
             "target.device_has_add": ["802_15_4_PHY"],
             "target.network-default-interface-type": "MESH"
@@ -32,7 +34,15 @@
         },
         "KW24D": {
             "LED": "LED1",
-            "BUTTON": "SW1"
+            "BUTTON": "SW1",
+            "mbed-mesh-api.heap-size": 20000,
+            "mcr20a.provide-default": true
+        },
+        "KW41Z": {
+            "LED": "LED1",
+            "BUTTON": "SW3",
+            "mbed-mesh-api.heap-size": 20000,
+            "mcr20a.provide-default": true
         },
         "NUCLEO_F401RE": {
             "LED": "NC",
@@ -52,4 +62,5 @@
             "BUTTON": "SW2"
         }
     }
+
 }

theotherjimmy, 2019-01-03:
>  Obviously there are a compiler/linker option getting dropped. 

Please don't assume that there is a bug before exhausting all other explications. `mbed compile` uses the `--profile develop` argument when none is specified. Please be sure to use the develop configuration after export. Perhaps you're trying to build with the equivalent of `--profile debug`, which may not fit.

jahess, 2019-01-09:
Sweet!  --profile profile worked.

Where is --profile documented?

It doesn't show up in "mbed [export|config|target] --help."  Perhaps my familiarity with the cli tutorials has become dated?

bridadan, 2019-01-09:
> It doesn't show up in "mbed [export|config|target] --help." Perhaps my familiarity with the cli tutorials has become dated?

It should be present for "mbed export", "mbed compile", and "mbed test". But I see you're right that it is missing from "mbed export", oops! We can definitely get that added.

theotherjimmy, 2019-01-09:
@bridadan When you add it please be sure to mention that it defaults to `debug`

jahess, 2019-01-09:
> When we add it please be sure to mention that it defaults to debug
>
> Edit: Let's not imply @bridadan <https://github.com/bridadan> more work.

Maybe it should default to whatever "mbed compile" defaults to so the newbies won't be immediately at a loss?

jahess, 2019-01-09:
I can also verify that the bin files are within 144 bytes of each other for:

mbed compile [-t GCC_ARM]
mbed export --profile mbed-os/tools/profiles/develop.json -i make_gcc_arm
mbed export --profile mbed-os/tools/profiles/develop.json -i gnuarmeclipse

Also, the bin's produce similar behavior on the target hardware with my limited testing.

closed 2019-01-09 by jahess