ARMmbed/mbed-cli #816: export -i make_gcc_arm build fails

jahess, 2019-01-03:
Exporting mbed-os-example-mesh-minimal with -i make_gcc_arm and -m KW24D results in a project that will not link.  The error from make is:

/Users/jahess/opt/gcc-arm-none-eabi-7-2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: mbed-os-example-mesh-minimal.elf section `.text' will not fit in region `m_text'
/Users/jahess/opt/gcc-arm-none-eabi-7-2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: region m_text overflowed with text and data
/Users/jahess/opt/gcc-arm-none-eabi-7-2018-q2-update/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: region `m_text' overflowed by 228812 bytes
collect2: error: ld returned 1 exit status
make: *** [mbed-os-example-mesh-minimal.elf] Error 1

The mbed_app.json was modified to work with the KW24D as shown below.  This same setup works with "mbed compile."  Obviously there are a compiler/linker option getting dropped.  The include directory list for individual files is also extremely long.

For the successful mbed compile the following global options were in affect:
GNU_ARM_PATH=/Users/jahess/opt/gcc-arm-none-eabi-6-2017-q2-update/bin
TOOLCHAIN=GCC_ARM

diff --git a/mbed_app.json b/mbed_app.json
index 8de7119..df32d68 100644
--- a/mbed_app.json
+++ b/mbed_app.json
@@ -1,5 +1,9 @@
 {
     "config": {
+        "storage-device":{
+            "help": "Values for storage (Thread) devices are: MESH_NVM_HEAP, MESH_NVM_SD_CARD and MESH_NVM_NONE",
+            "value": "MESH_NVM_NONE"
+        },
         "enable-led-control-example": true,
         "LED": "NC",
         "BUTTON": "NC",
@@ -7,17 +11,15 @@
     },
     "target_overrides": {
         "*": {
-            "nanostack.configuration": "lowpan_router",
-            "nsapi.default-mesh-type": "LOWPAN",
-            "mbed-mesh-api.6lowpan-nd-panid-filter": "0xffff",
-            "mbed-mesh-api.6lowpan-nd-channel-page": 0,
-            "mbed-mesh-api.6lowpan-nd-channel": 12,
-            "mbed-mesh-api.6lowpan-nd-channel-mask": "(1<<12)",
-            "mbed-mesh-api.heap-size": 14000,
+            "nanostack.configuration": "thread_router",
+            "nsapi.default-mesh-type": "THREAD",
             "mbed-trace.enable": false,
+            "mbed-mesh-api.heap-size": 30000,
+            "mbed-mesh-api.thread-device-type": "MESH_DEVICE_TYPE_THREAD_ROUTER",
             "platform.stdio-convert-newlines": true,
             "platform.stdio-baud-rate": 115200,
-            "atmel-rf.provide-default": true,
+            "mbed-mesh-api.thread-use-static-link-config": true,
+            "atmel-rf.provide-default": false,
             "mcr20a.provide-default": false,
             "target.device_has_add": ["802_15_4_PHY"],
             "target.network-default-interface-type": "MESH"
@@ -32,7 +34,15 @@
         },
         "KW24D": {
             "LED": "LED1",
-            "BUTTON": "SW1"
+            "BUTTON": "SW1",
+            "mbed-mesh-api.heap-size": 20000,
+            "mcr20a.provide-default": true
+        },
+        "KW41Z": {
+            "LED": "LED1",
+            "BUTTON": "SW3",
+            "mbed-mesh-api.heap-size": 20000,
+            "mcr20a.provide-default": true
         },
         "NUCLEO_F401RE": {
             "LED": "NC",
@@ -52,4 +62,5 @@
             "BUTTON": "SW2"
         }
     }
+
 }

0xc0170, 2019-01-09:
Is this related to mbed-cli or rather Mbed OS ? Should it be reported there, as it seems some configuration is missing for a target?

This is similar target, being added just now https://github.com/ARMmbed/mbed-os/pull/6622

theotherjimmy, 2019-01-09:
@jahess The default `--profile` argument differs between `mbed compile` and `mbed export` can you confirm that explicitly setting `--profile` does not produce the same result?

jahess, 2019-01-09:
Sweet!  --profile develop profile worked.

Where is --profile documented?

It doesn't show up in "mbed [export|config|target] --help."  Perhaps my familiarity with the cli tutorials has become dated?

jahess, 2019-01-09:
@theotherjimmy,

When you ask if they produce the same results, what do you mean?

$ cmp BUILD/mbed-os-example-mesh-minimal.bin BUILD/KW24D/GCC_ARM/mbed-os-example-mesh-minimal.bin 
BUILD/mbed-os-example-mesh-minimal.bin BUILD/KW24D/GCC_ARM/mbed-os-example-mesh-minimal.bin differ: char 1121, line 1

Says are different.  I looked deeper and there are many differences.

theotherjimmy, 2019-01-09:
> When you ask if they produce the same results, what do you mean?

Mostly that they both build the code and consume about the same space.

jahess, 2019-01-09:
They were within 50 bytes of each other.  Also, I have verified that bins execute on the boards the same in my limited testing.

closed 2019-01-09 by jahess