ARMmbed/mbed-cli #817: The mbed tools were not found

dinglong-akm, 2019-01-08:
I use command "mbed import https://github.com/.../Popup.git -v" to get source code, but it failed to git clone mbed-os, so i use "git clone git@github.com:ARMmbed/mbed-os.git" to get and put it.
     but when i run command "mbed compile -m NUCLEO_F411RE -t GCC_ARM -v", it tips 

D:\work_src\akm_src\compass\12_18_popup\Popup>mbed compile -m NUCLEO_F411RE -v
[mbed] Working path "D:\work_src\akm_src\compass\12_18_popup\Popup" (library)
[mbed] Program path "D:\work_src\akm_src\compass\12_18_popup\Popup"
[mbed] ERROR: The mbed tools were not found in "D:\work_src\akm_src\compass\12_18_popup\Popup".
       You can run "mbed deploy" to install dependencies and tools.
---
    
    Do you know how to deal with it???

theotherjimmy, 2019-01-08:
@dinglong-akm Please be sure to clone mbed-os into the project in the `mbed-os` directory.

theotherjimmy, 2019-01-08:
If you have done that, could you post your project structure here?

dinglong-akm, 2019-01-09:
i know the method to deal with it, just need set environment "mbed config -G MBED_OS_DIR .\mbed-os".

theotherjimmy, 2019-01-09:
@dinglong-akm That should not be necessary, as Mbed CLI already looks for mbed-os in `./mbed-os`. Further, that configuration is probably not going to do what you think: it will not resolve that path, and will always assume that mbed-os is relative to the location you execute `mbed ...` from.

naveenkaje, 2019-03-08:
Hi @dinglong-akm 
mbed-os needs to be included as a dependency in your project. Please look at https://github.com/ARMmbed/mbed-os-example-blinky/blob/master/mbed-os.lib as an example to see how mbed-os is being referenced from the mbed-os-example-blinky project.

With this dependency, you can see the following behavior. Please let us know if that resolves your issue.

```
/c/
$ mbed import https://github.com/ARMmbed/mbed-os-example-blinky -v
Cloning into 'C:\mbed-cli\new_import_test\mbed-os-example-blinky'...
remote: Enumerating objects: 591, done.
remote: Total 591 (delta 0), reused 0 (delta 0), pack-reused 591
Receiving objects: 100% (591/591), 258.96 KiB | 2.18 MiB/s, done.
Resolving deltas: 100% (178/178), done.
Fetching origin
remote: Enumerating objects: 5463, done.
remote: Counting objects: 100% (5463/5463), done.
remote: Total 11208 (delta 5462), reused 5462 (delta 5462), pack-reused 5745
Receiving objects: 100% (11208/11208), 15.73 MiB | 13.25 MiB/s, done.
Resolving deltas: 100% (8235/8235), completed with 1811 local objects.
From https://github.com/ARMmbed/mbed-os
 + ed072b0ec6...45c727ee6f feature-armc6      -> origin/feature-armc6  (forced update)
 + 34c933e917...e64286be90 feature-iar8       -> origin/feature-iar8  (forced update)
 + dc1979a7dd...7e5833e27b feature-watchdog   -> origin/feature-watchdog  (forced update)
 + 7d8ec83551...7098f1b7f9 fh_enable          -> origin/fh_enable  (forced update)
   baed84a109..b582c54a5f  master             -> origin/master
   bc132a6066..51d55508e8  mbed-os-5.11       -> origin/mbed-os-5.11
 * [new branch]            mbed-os-5.12       -> origin/mbed-os-5.12
   36734f4fb5..b83fb2b738  release-candidate  -> origin/release-candidate
 t [tag update]            latest             -> latest
 * [new tag]               mbed-os-5.11.5     -> mbed-os-5.11.5
 * [new tag]               mbed-os-5.12.0-rc1 -> mbed-os-5.12.0-rc1
 * [new tag]               mbed_lib_rev165    -> mbed_lib_rev165
[mbed] Working path "C:\mbed-cli\new_import_test" (directory)
[mbed] Importing program "mbed-os-example-blinky" from "https://github.com/ARMmbed/mbed-os-example-blinky" at latest revision in the current branch
[mbed] Trying to guess source control management tool. Supported SCMs: bld, git, hg
[mbed] Exec "git clone https://github.com/ARMmbed/mbed-os-example-blinky C:\mbed-cli\new_import_test\mbed-os-example-blinky" in "C:\mbed-cli\new_import_test"
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #51d55508e840
[mbed] Trying to guess source control management tool. Supported SCMs: bld, git, hg
[mbed] Found matching cached repository in "C:\Users\navkaj01\.mbed\mbed-cache\github.com\ARMmbed/mbed-os"
[mbed] Carbon copy from "C:\Users\navkaj01\.mbed\mbed-cache\github.com\ARMmbed/mbed-os" to "C:\mbed-cli\new_import_test\mbed-os-example-blinky\mbed-os"
[mbed] Setting url to "https://github.com/ARMmbed/mbed-os" in C:\mbed-cli\new_import_test\mbed-os-example-blinky\mbed-os
[mbed] Cleaning up Git index
[mbed] Update cached copy from remote repository
[mbed] Fetching revisions from remote repository to "mbed-os"
[mbed] Exec "git fetch --all --tags" in "C:\mbed-cli\new_import_test\mbed-os-example-blinky\mbed-os"
[mbed] Discarding local changes in "mbed-os"
[mbed] Checkout "51d55508e8400b60af467005646c4e2164738d48" in mbed-os
[mbed] Revision "51d55508e8400b60af467005646c4e2164738d48" matches a branch "mbed-os-5.11" reference. Re-associating with branch
[mbed] Exec "git checkout mbed-os-5.11 -q" in "C:\mbed-cli\new_import_test\mbed-os-example-blinky\mbed-os"
[mbed] Auto-installing missing Python modules...

$ ls /c/mbed-os-example-blinky
CONTRIBUTING.md  main.cpp  mbed_app.json  mbed_settings.py  mbed-os/  mbed-os.lib  README.md  stats_report.h
```

mark-edgeworth, 2019-06-25:
@dinglong-akm It looks like this issue has been resolved now, so I'm going to close it. Please feel free to reopen with further information if you need to do so.

closed 2019-06-25 by mark-edgeworth