ARMmbed/mbed-cli #819: How to get rid of the error tip ?

prowan, 2019-01-09:
I have exported a project to eclipse, and then set the Build command in eclipse as follow:

mbed-cli compile -j0 -t GCC_ARM -m NUCLEO_F767ZI --profile mbed-os/tools/profiles/debug.json

This works well, but how to get rid of the error tip in eclipse as illustrated in this picture, thank a lot !
[image]

theotherjimmy, 2019-01-11:
@prowan Thanks for including the paths that you added. Those are pretty specific to your install of gcc, including the version. I wonder if there is a way to get eclipse to add those for us.

linlingao, 2019-05-29:
It appears your issue has been resolved. We'll be closing this.

closed 2019-05-31 by ciarmcom