ARMmbed/mbed-cli #825: Compiles to .hex and not to .bin

Akselvb, 2019-01-15:
Hello!

When trying to compile, the following happens:

[image]

The resulting build is compiled as a .hex file instead of a .bin file. However, an _Update Image_ is created as a .bin file. This happens for all projects. Is it correct output?

teetak01, 2019-01-15:
Hi @Akselvb, yes this is expected output. The `Image` is the one you should flash to your board. `Update image` is used by Device Management update feature, and that only works with .bin format images, thus the default output there is .bin. You can also override this to output  also the update image as .hex (or something else) if you need the update image to be of different format.

Is this an issue for you, or are you just wondering why this changed?

The change was done to have more streamlined support for Pelion device management update when compiling the application firmware image in .hex format. This way you do not need to edit the mbed_app.json to recompile the whole application to get updatable image in .bin format.

Akselvb, 2019-01-15:
Hi @teetak01, thank you for answering. It is no issue, more than a confusion whether that was the reason my compilation is failing. Only seeing .bin files as Images, I thought it might be the reason for the following compilation error:

[image]

I thought the generated .hex file maybe was bigger than the corresponding .bin, causing the overflow.

However, when I change target from _NRF52_DK_ to _EFM32GG11_STK3701_ (larger flash), it compiles and I get a .bin file:

[image]

Is the format of the compiled image MCU-dependent?

Akselvb, 2019-01-15:
The last link is dead.

0xc0170, 2019-01-15:
> Internal Jira reference: https://jira.arm.com/browse/MBOCUSTRIA-486

@Akselvb This is internal ticket reference to keep track of the external issues internally.

theotherjimmy, 2019-01-15:
> Is the format of the compiled image MCU-dependent?

Yes.

See https://os.mbed.com/docs/mbed-os/v5.11/reference/adding-and-configuring-targets.html, `OUTPUT_EXT` section.

closed 2019-01-20 by Akselvb