ARMmbed/mbed-cli #826: Mbed import with `--depth` switch should affect mbed-os clone

theotherjimmy, 2019-01-18:
from @fredlee12001 

> mbed import accepts an option to truncate the git history depth, `--depth`.
When I tried to import the PDMC project with `--depth 1`, I found that Mbed OS contained more than one commit in its history. See attached image
[image]

bridadan, 2019-01-18:
@theotherjimmy I believe this comes from the fact that you can't clone a specific commit with git. And since we pin specific commits with the `.lib` files, you usually don't get the commit you need if you only clone one of them.

That being said, I was wondering if it would work with _tags_: `git clone -b "mbed-os-5.10.2" --depth 1 https://github.com/armmbed/mbed-os`

That only gave me a history of one commit and quite a small `.git` folder (48.5 MB)

screamerbg, 2019-02-01:
This might be also related to locally cached copy of Mbed OS that has the full index tree. To save traffic, Mbed CLI will prefer to use the locally cached copy (200KB download vs 50MB)

closed 2019-10-31 by ciarmcom