ARMmbed/mbed-cli #827: mbed-cli Manual installation with python3

kimsooyoung, 2019-01-21:
mbed-cli Manual installation with python3
===========================

I follwed manual installation of mbed - cli on windows 
> <https://os.mbed.com/docs/mbed-os/v5.11/tools/windows.html>

Here's my env
----------------------
>windows 10
>python 3.6.6 (same for pip)
>mbed cli 1.8.3
>gcc compiler 8 2018-q4-major
>mbed os - more than 5.9 (actually don't know -- was my first time)
>TARGET "DISCO_L475VG_IOT01A"

Just followed manual && got some troubles && shoot them 
Actually, I modified python source code a little, For mbed users with python3, I'll leave my footprints
I don't know whether such problem is just for me. Anyway, for sbd who likes me, let's start ;) 

# 1. install problem 
if you imported example and type 
<pre><code>$ mbed compile -c</code></pre>
it starts to install python libraries automatically and maybe it consider dependencies(maybe... I'm just stranger about this) So, I got some errors like **egg~~** something (I forgot to capture that just remember egg 👍 ) 
If you open mbed-os file, there's list of require things *(requirement.txt)*
I tried stupid, but definite way
<pre><code>$ pip install "name of library one-by-one"</code></pre>
&& No such commands like need ~~ library occurs 👍 
So much Heuristic, Anyway, && typed such commands in my example folder
<pre><code>$ mbed config -G GCC_ARM_PATH "(folder that my ones - toolchain exist )~\bin"
$ mbed config -G TOOLCHAIN "GCC_ARM"</code></pre>
than, typed compile again. && Got this Error
<pre><code>[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "C:\Users\tge13\workshop\mbed-os-example-blinky" (program)
Traceback (most recent call last):
  File "C:\Users\tge13\workshop\mbed-os-example-blinky\mbed-os\tools\make.py", line 40, in <module>
    from tools.options import get_default_options_parser
  File "C:\Users\tge13\workshop\mbed-os-example-blinky\mbed-os\tools\options.py", line 24, in <module>
    from .toolchains import TOOLCHAINS
  File "C:\Users\tge13\workshop\mbed-os-example-blinky\mbed-os\tools\toolchains\__init__.py", line 43, in <module>
    from ..config import (ConfigException, RAM_ALL_MEMORIES, ROM_ALL_MEMORIES)
  File "C:\Users\tge13\workshop\mbed-os-example-blinky\mbed-os\tools\config\__init__.py", line 37, in <module>
    from ..arm_pack_manager import Cache
  File "C:\Users\tge13\workshop\mbed-os-example-blinky\mbed-os\tools\arm_pack_manager\__init__.py", line 2, in <module>
    from urllib2 import urlopen, URLError
  File "c:\program files (x86)\microsoft visual studio\shared\python36_64\lib\site-packages\urllib2.py", line 220
    raise AttributeError, attr
                        ^
SyntaxError: invalid syntax
[mbed] ERROR: "c:\program files (x86)\microsoft visual studio\shared\python36_64\python.exe" returned error.
       Code: 1
       Path: "C:\Users\tge13\workshop\mbed-os-example-blinky"
       Command: "c:\program files (x86)\microsoft visual studio\shared\python36_64\python.exe -u C:\Users\tge13\workshop\mbed-os-example-blinky\mbed-os\tools\make.py -t GCC_ARM -m DISCO_L475VG_IOT01A --source . --build .\BUILD\DISCO_L475VG_IOT01A\GCC_ARM -c"
       Tip: You could retry the last command with "-v" flag for verbose output
---
[mbed] WARNING: Using Python 3 with Mbed OS 5.8 and earlier can cause errors with compiling, testing and exporting
--- </code></pre>

Was, **SyntaxError** And, opened __init__.py. It starts like 

<pre><code>try:
	from urllib2 import urlopen, URLError    
except ImportError:
    from urllib.request import urlopen, URLError
</code></pre>

urllib2 && raise AttributeError, attr... such syntax are for python2, So I swapped two lines,

<pre><code>try:
    from urllib.request import urlopen, URLError
except ImportError:
    from urllib2 import urlopen, URLError
</code></pre>

Then I succeed in doing compile
[image]
mbed-cli users with python3, If you got such problem, try this, I wonder It's just for me or not.

screamerbg, 2019-01-31:
@cmonr FYI. Please investigate.

cmonr, 2019-01-31:
@kimsooyoung A couple of things:

1) Could you provide the output for what you get when you run `pip list`?
2) > mbed os - more than 5.9 (actually don't know -- was my first time)
Please provide the vesrion of mbed-os you're using, as this is where the requirements.txt for the tools lives (to this day, I still find it a bit odd)
3) > gcc compiler 8 2018-q4-major
GCC 8 isn't supported as of yet, even in the latest Mbed OS. As of right now, we support GCC 6, and have heard of successful usage with GCC 7 (we're working on updating our compilers)

cmonr, 2019-02-03:
@kimsooyoung Thanks for the outputs. A couple of things.

> 2. Got some problems about pip
3. Solved that, By install everything one-by-one, Not using requirement.txt.

I still don't understand why you're installing everything manually instead of using `pip install -r requirements.txt`.

Part of the reason we provide the file to begin with is that because of the large number of modules that we use with Mbed OS tools, numerous interdependent bugs can and have appeared, and we use requirements.txt to retain a known working set of versions with Mbed OS. We try to be as lax as we cam when it comes to version restriction, but at the same time trying to support all versions of all modules would be incredibly manigable.

In this particular issue, you're coming across a specific bug that I don't think anyone has seen this far, and are using modules that don't match what we test with. Truth be told, I'm not surprised you were hitting problems earlier.

I would be interested in seeing what happens if you downgrade both urllib3 and requests to the versions specified in `requiremets.txt`, and seeing if you still need to modify init.py. 

Another thing you could try is instead of swapping the two lines, you might look into modifying the except line as such:
`except ImportError, AttributeError:`
The intention with that block of code was to attempt the import if Py2 was installed, and if Py3 was instead being used, catch the error we knew would be thrown, and do the import for Py3.

Finally, the list of modules that you have installed is _really_ long. You could use a tool called `pipenv` to create and manage a virtual environment just for Mbed OS. This could help keep your system modules (and their specific versions) seperate from the ones needed and tested against Mbed OS.

bridadan, 2019-02-04:
Hi @kimsooyoung, thanks for posting that issue regarding the pyocd install issue, I believe I've found and fixed the root cause of that issue here: https://github.com/mbedmicro/pyOCD/pull/528. I think that particular error may be dependent on what character set your terminal is using, so that might explain why I don't see that issue when installing pyocd.

Sounds like you are able to now use mbed-cli after using `pip install -r requirements.txt`. We're looking to make the installation of these dependencies easier in the future as other people are have similar issues. But for the time being, are you able to use Mbed CLI?

kimsooyoung, 2019-02-05:
I saw the issue about pyOCD's readme.md problem, And I understand about why cannot install pyOCD v0.14 I'm Korean and Korean character encoding type is cp949 - extension of EUC-KR

Thx so much I learned about encoding type and Don't need to modify init python file 
But, In order to use requirements.txt I modified that text file 

`pyocd==0.16`

And... got this message 
`pyocd 0.16.0 has requirement intervaltree<4.0,>=3.0.2, but you'll have intervaltree 2.1.0 which is incompatible.`

> In Eclipse, select the "Help -> Install New Software…" menu item. Then either click the "Add…"

> In Eclipse, select the "Help -> Install New Software..." menu item. Then either click the "Add..."

bridadan, 2019-02-05:
Thanks very much! Sorry about the pip issues, its kind of the nature of the tool. We'd like to improve this process in the future though!

closed 2019-02-05 by bridadan

cmonr, 2019-02-11:
@kimsooyoung I'm glad you and @bridadan were able to resolve this while I was out.

> But, I got such error when install pyocd. In fact, I participated in Mbed-OS workshop in South Korea.
During workshop, some people got same error as me. That's the reason why I didn't use requirements.

That makes a lot of sense. There was a recent window of time where we had a bout of module incompatabilities that caused installations to fail which has been clamped on. 

Imo, the best that we can do right now is guarantee that installation always works within a set of module versions, not necessarily _all_ versions of _all_ modules.

Glad you were able to get it sorted out, and I appologize if my last response sounded a bit terse.

kimsooyoung, 2019-02-13:
No problem. That's okay :) Thx for your answer, I could solve my problem thx to your advise.
And `thank you` for respect my opinion.