ARMmbed/mbed-cli #832: mbed cli autoinstall hangs with pip 19.0.1

JanneKiiskila, 2019-01-31:
If you upgrade pip to version 19.0.1 (latest as of today for Ubuntu 16.04 at least).

```
usr@ubuntu:~/mbed/mbed-cloud-client-example-internal$ pip --version
pip 19.0.1 from /usr/local/lib/python2.7/dist-packages/pip (python 2.7)

causes `mbed compile` to fail, as it it ends up in some never-ending loop inside the auto-installer.
```
usr@ubuntu:~/mbed/mbed-cloud-client-example-internal$ mbed compile -m K66F -t GCC_ARM --stats-depth 100
[mbed] Working path "/home/jankii01/mbed/mbed-cloud-client-example-internal" (program)
[mbed] Auto-installing missing Python modules (intervaltree, mbed_cloud_sdk, icetea)...
[mbed] Auto-installing missing Python modules (intervaltree, mbed_cloud_sdk, icetea)...
^C[mbed] User aborted!
```
(after 20 minutes or so).

Actually, it's likely all the `mbed` commands that evoke auto-install,  same happens also with `mbed deploy`.

JanneKiiskila, 2019-01-31:
@neilwjackson  @theotherjimmy @SenRamakri

JanneKiiskila, 2019-01-31:
You can downgrade back with `pip install pip==18.1`, it starts to work again after that.

screamerbg, 2019-01-31:
@JanneKiiskila I'm unable to reproduce the problem with pip 19.0.1, Mbed CLI 1.8.3 and https://github.com/ARMmbed/mbed-cloud-client-example-internal

Here's my log:
```
D:\1>mbed import https://github.com/ARMmbed/mbed-cloud-client-example-internal
[mbed] Working path "D:\1" (directory)
[mbed] Program path "D:\1"
[mbed] Importing program "mbed-cloud-client-example-internal" from "https://github.com/ARMmbed/mbed-cloud-client-example-internal" at latest revision in the current branch
[mbed] Adding library "mbed-cloud-client" from "https://github.com/ARMmbed/mbed-cloud-client-internal" at latest revision in the current branch
[mbed] Adding library "mbed-cloud-client\certificate-enrollment-client" from "https://github.com/ARMmbed/certificate-enrollment-client" at rev #949c358bee9c
[mbed] Adding library "mbed-cloud-client\factory-configurator-client" from "https://github.com/ARMmbed/factory-configurator-client" at rev #bca7efbc1c40
[mbed] Adding library "mbed-cloud-client\factory-configurator-client\mbed-client-esfs" from "https://github.com/ARMmbed/mbed-client-esfs" at rev #0e1429c6c919
[mbed] Adding library "mbed-cloud-client\factory-configurator-client\secsrv-cbor" from "https://github.com/ARMmbed/secsrv-cbor" at rev #37d8d135604c
[mbed] Adding library "mbed-cloud-client\mbed-client-pal" from "https://github.com/ARMmbed/mbed-client-pal" at rev #9d3ebaaefb18
[mbed] Adding library "mbed-cloud-client\mbed-client-randlib" from "https://github.com/ARMmbed/mbed-client-randlib" at rev #531457e3b095
[mbed] Adding library "mbed-cloud-client\mbed-coap" from "https://github.com/ARMmbed/mbed-coap" at rev #7cc17aaa5f85
[mbed] Adding library "mbed-cloud-client\mbed-trace" from "https://github.com/ARMmbed/mbed-trace" at rev #6d3590f08853
[mbed] Adding library "mbed-cloud-client\nanostack-libservice" from "https://github.com/ARMmbed/nanostack-libservice" at rev #a953636946a4
[mbed] Adding library "mbed-cloud-client\ns-hal-pal" from "https://github.com/ARMmbed/ns-hal-pal" at rev #df53a563e067
[mbed] Adding library "mbed-cloud-client\sal-stack-nanostack-eventloop" from "https://github.com/ARMmbed/sal-stack-nanostack-eventloop" at rev #be7cdd6bf4b1
[mbed] Adding library "mbed-cloud-client\update-client-hub" from "https://github.com/ARMmbed/update-client-hub" at rev #cdbabf8116a5
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #cee814dc2971
[mbed] Adding library "pal-platform" from "https://github.com/ARMmbed/pal-platform" at rev #bc8a1d84d7bf
[mbed] Adding library "tools" from "https://github.com/ARMmbed/mbed-bootloader-binaries" at rev #6b5c757abc29
[mbed] Adding library "drivers\wifi-x-nucleo-idw01m1" from "https://github.com/ARMmbed/wifi-x-nucleo-idw01m1" at rev #a00282000178
[mbed] Adding library "source\platform" from "https://github.com/ARMmbed/mbed-cloud-client-platform-common" at rev #eb51794b3fee
[mbed] Auto-installing missing Python modules (intervaltree, mbed_cloud_sdk, icetea)...

D:\1>mbed --version
1.8.3

D:\1>pip --version
pip 19.0.1 from c:\progra~1\python\lib\site-packages\pip (python 2.7)
```

Can you provide:
* Mbed CLI version
* Python version

JanneKiiskila, 2019-01-31:
It's now some other dependency that is causing it, @anttiylitokola also has it working with 19.0.1. So, there's now some other combination that causes the issue here - downgrading fixed the issue for me.

JanneKiiskila, 2019-02-01:
I am probably also having a newer version of `git`:
```
git --version
git version 2.20.1
```

bridadan, 2019-03-04:
@JanneKiiskila this should have been fixed with release 1.9.0, are you still experiencing this issue?

JanneKiiskila, 2019-03-25:
This can be closed now, closing it.

closed 2019-03-25 by JanneKiiskila