ARMmbed/mbed-cli #836: mbed-cli doesn't work with python3 on Ubuntu 18.04

yogpan01, 2019-02-01:
Ubuntu 18.04 now comes by default with python3 support and pip support is also through python3.
Using mbed-cli is now giving warning and sometimes weird failures failing to link the application because of some incompatible python code.

For example, I am trying to compile an application which requires to be built for ARMC6 but it fails in linking for me, even though I am using newer than 5.8 version of MbedOS.
```
 xxxx@ubuntu:~/example(master)$ mbed compile -t ARMC6 -m NUMAKER_PFM_M2351 --app-config configs/wifi_esp8266_v4.json
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "/home/yogpan01/example/mbed-client-example" (library)
[mbed] Program path "/home/yogpan01/example/mbed-client-example"
Building project mbed-client-example (NUMAKER_PFM_M2351, ARMC6)
Scan: mbed-client-lite-example-restricted
Link: mbed-client-lite-example-restricted
b"Warning: L3912W: Option 'legacyalign' is deprecated."
[ERROR] can't concat str to bytes
[mbed] ERROR: "/usr/bin/python3" returned error.
       Code: 1
       Path: "/home/xxxx/example/mbed-client-example"
       Command: "/usr/bin/python3 -u /home/yogpan01/example/mbed-client-example/mbed-os/tools/make.py -t ARMC6 -m NUMAKER_PFM_M2351 --source . --build ./BUILD/NUMAKER_PFM_M2351/ARMC6 --app-config configs/wifi_esp8266_v4.json"
       Tip: You could retry the last command with "-v" flag for verbose output
---
[mbed] WARNING: Using Python 3 with Mbed OS 5.8 and earlier can cause errors with compiling, testing and exporting
---
```

screamerbg, 2019-02-01:
@yogpan01 Thanks for the bug report

cmonr, 2019-02-01:
@yogpan01 Cna you provide which version of Mbed OS you're using, as well the output of `pip list`?

cmonr, 2019-02-01:
Also, as a rule of thumb, using verbose output when debugging is generally really helpful.

Would appreciate seeing the more verbose log.

cmonr, 2019-02-13:
@yogpan01 @screamerbg Please retest this with a fresh Mbed OS master branch pull.

https://github.com/ARMmbed/mbed-os/pull/9673 came in today which fixed a parsing error when using Py3 that looks like it was also affecting this PR. 

If not, please provide more verbose output by also using the `-v` flag.

yogpan01, 2019-02-19:
It works fine on MbedOS master, when we can have this fix for any mbed-os official release?

vmedcy, 2019-02-19:
@yogpan01: ARMmbed/mbed-os#9673 targets 5.11.5 which is planned for the next week I guess.

cmonr, 2019-02-19:
@yogpan01 My mistake. Also saw the internal comment. Will keep open until 5.11.5 RC PR is created and merged.

cmonr, 2019-02-28:
Closing since 5.11.5 is now released.

closed 2019-02-28 by cmonr