ARMmbed/mbed-cli #839: Readme / Wiki pages broken links

marcin-ch, 2019-02-04:
I believe it refers exclusively to `README.md`, links like [Creating and importing programs](https://github.com/ARMmbed/mbed-cli#creating-and-importing-programs) and [Importing an existing program](https://github.com/ARMmbed/mbed-cli#importing-an-existing-program) seem to be broken.

Could you please review it.

theotherjimmy, 2019-02-04:
Which readme are you referring to? The README in the root simply refers to the documentation at os.mbed.com/docs

naveenkaje, 2019-02-04:
@marcin-ch As @theotherjimmy pointed out, https://github.com/ARMmbed/mbed-cli/blob/master/README.md links to https://os.mbed.com/docs/mbed-os/v5.11/tools/developing-mbed-cli.html -> https://github.com/ARMmbed/mbed-cli/blob/1.8.3/README.md#creating-and-importing-programs/https://github.com/ARMmbed/mbed-cli/blob/1.8.3/README.md#importing-an-existing-program. Can you please let us know which URLs don't work for you?

marcin-ch, 2019-02-05:
@theotherjimmy @naveenkaje I have some links saved, and they are for example as follows (and mentioned in my first post):
[Creating and importing programs](https://github.com/ARMmbed/mbed-cli#creating-and-importing-programs) = `https://github.com/ARMmbed/mbed-cli#creating-and-importing-programs`
[Importing an existing program](https://github.com/ARMmbed/mbed-cli#importing-an-existing-program) = `https://github.com/ARMmbed/mbed-cli#importing-an-existing-program`

So as you can see, they are different than you pointed out.
I am pretty sure that above links referred to instructions available under `README.md`. It looked like below in the picture (it has been prepared in Paint):
[image]

So, did it change recently? I am pretty sure I did save links in December '18 and they worked then just fine. Now, I can see only [Introduction](https://github.com/ARMmbed/mbed-cli#introduction)

Hope that is clear now.

marcin-ch, 2019-02-06:
Now I see here in online docs, section [Developing: Mbed CLI](https://os.mbed.com/docs/mbed-os/latest/tools/developing-mbed-cli.html) the note that:
**You can find our legacy CLI documentation in the [README](https://github.com/ARMmbed/mbed-cli/blob/1.8.3/README.md).**

@theotherjimmy @naveenkaje @bridadan thanks then for all the explanation and your support!

closed 2019-02-06 by marcin-ch