ARMmbed/mbed-cli #843: mbed dm update device cannot delete campaign correctly if interrupted

LiyouZhou, 2019-02-13:
mbed dm update device will start a campaign and wait for campaign to go into autostopped state, but if I interrupt this process, mbed dm will try to delete a campagin in publishing mode which is not allowed. Hence I am greeted with error:
```
[INFO] 2019-02-13 12:42:58 - manifesttool.update_device - Campaign successfully started. Current state: u'scheduled'. Checking updates..
[INFO] 2019-02-13 12:42:59 - manifesttool.update_device - Current state: 'checkedmanifest'
[INFO] 2019-02-13 12:43:00 - manifesttool.update_device - Current state: 'publishing'
^C[mbed] User aborted!
[CRITICAL] 2019-02-13 12:43:24 - manifesttool.update_device - User Aborted... Cleaning up.
[INFO] 2019-02-13 12:43:24 - manifesttool.update_device - ** Deleting update campaign and manifest **

~/work/mbed-cloud-client-example-internal master* 30s
❯ [CRITICAL] 2019-02-13 12:43:24 - manifesttool.update_device - Cleanup of campaign failed with:
(409) "Conflict" (409)
Reason: Conflict
HTTP response headers: HTTPHeaderDict({'Content-Length': '162', 'Access-Control-Expose-Headers': 'X-Request-ID,Content-Length,Content-Disposition', 'Set-Cookie': 'TS017f188d=011138c4dfd0d2dc94d268100a05c08eeeccb1e96fe84e7b5d33548f772aa4c892cd3601c6230cfdac06dbc1af23712a18cb5cd8d7; Path=/', 'Strict-Transport-Security': 'max-age=10886400; includeSubDomains; preload', 'Vary': 'Origin', 'X-Request-ID': '00005c6410ec4e4ac20ea9350000000a', 'Connection': 'keep-alive', 'Allow': 'GET, PUT, PATCH, DELETE, HEAD, OPTIONS', 'Date': 'Wed, 13 Feb 2019 12:43:24 GMT', 'Access-Control-Allow-Origin': '*', 'Content-Type': 'application/json'})
HTTP response body: {"object":"error","code":409,"type":"conflict","message":"Cannot delete the campaign while in the current phase.","request_id":"00005c6410ec4e4ac20ea9350000000a"}
```

The consequence of this is the campaign is left running, and the device is forever prevented from joining another campaign. Now I have to go into portal to manually stop the campaign and deleting it before i can call mbed dm again.

Solution is: to clean up a campaign, mbed dm should first stop it before deleting.

screamerbg, 2019-02-13:
We're experiecing the same issue when running the Pelion DM tests, which make use of `mbed dm` -   https://github.com/ARMmbed/simple-mbed-cloud-client/#testing

CC @theotherjimmy @bridadan

bridadan, 2019-02-13:
Thanks very much for providing the log and a proposed solution. What you've suggested makes sense to me. This part of the update process is handled by the [manifest-tool](https://github.com/ARMmbed/manifest-tool), so the fix would need to be handled by that tool and its team.

In fact, it looks like someone has already raised an issue for this problem: https://github.com/ARMmbed/manifest-tool/issues/7. Shall we close this issue and keep the conversation going over there?

LiyouZhou, 2019-02-13:
Issues on the manifest-tool are not monitored. I think it is best to keep this open.

teetak01, 2019-03-13:
Here is a fix for the SDK to enable this: https://github.com/ARMmbed/mbed-cloud-sdk-python/pull/322

After that manifest-tool could enable that by default.

closed 2019-05-07 by ciarmcom