ARMmbed/mbed-cli #849: Feature request: It would be nice to be able to comment configuration files, especially mbed_lib files

LiyouZhou, 2019-02-14:
This will keep the documentation at the point of use, and deliver a much better ux.

The limitation of Json can be by-passed if we used some format that supports comments:
https://hjson.org/
https://yaml.org/
This new format support does not need to break json support. they can co-exist. mbed-cli will parse the file according to its file extension.

loverdeg-ep, 2019-02-14:
Maybe this can become the start of a real specification for mbed_libs?

Either way I'd like dependency/package management to be handled by a more modular solution or entirely different tool.

theotherjimmy, 2019-02-14:
@LiyouZhou This is an issue for mbed-os.

LiyouZhou, 2019-02-14:
@LiyouZhou You have admin access to this repo, can you transfer this issue please? https://help.github.com/articles/transferring-an-issue-to-another-repository/

mark-edgeworth, 2019-06-25:
As for the 'real spec for libraries'; this is a really important request but should be a new issue. We already have it captured internally as https://jira.arm.com/browse/IOTBTOOL-312. 
@LiyouZhou I, too, am irritated by the lack of comment in JSON, but this is unlikely to be fixed in the medium term as we'd have to convert all of our configs to YAML or something like it. I'm closing this original issue as won't fix.

closed 2019-06-25 by mark-edgeworth