ARMmbed/mbed-cli #854: Remove the obsolete python3 warning from mbed-cli

teetak01, 2019-02-18:
This warning should be removed from the next version.

[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting

Its printed with every mbed command regardless which Mbed OS I am using.

teetak01, 2019-02-18:
The warning is not really even correct as not even 5.11.4 works with python3.

naveenkaje, 2019-02-19:
@teetak01 Thanks for reporting this issue.

GitHub Issue Administration Note: Somehow the labels on this issue can not be edited for triage/tracking purpose. Needs to be investigated.

bridadan, 2019-02-19:
I've also received this warning twice in the same operation. In this case, I mistakenly tried to reimport a project:

```
$ mbed import <project url>
C:\Users\bridan01\onedrive_arm\Documents\dev>mbed import https://github.com/ARMmbed/mbed-cloud-client-example-internal
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "C:\Users\bridan01\onedrive_arm\Documents\dev" (directory)
[mbed] ERROR: Directory "C:\Users\bridan01\onedrive_arm\Documents\dev\mbed-cloud-client-example-internal" is not empty. Please ensure that the destination folder is empty.
---
[mbed] WARNING: Using Python 3 with Mbed OS 5.8 and earlier can cause errors with compiling, testing and exporting

mark-edgeworth, 2019-06-13:
https://github.com/ARMmbed/mbed-cli/pull/900 created

closed 2019-06-19 by ciarmcom