ARMmbed/mbed-cli #855: Added tests in UNITTESTS directory but not recognised with 'mbed status' command

ZaneV, 2019-02-21:
First time posting a github issue and I feel like this is a rookie error I am making but here goes:

I have added unit test files under the mbed-os/UNITTESTS directory and am at the stage that I want to commit all my first and push. However, (I am using Mercurial) if I use the command :
`mbed status`
No changes have been detected.
Also hg add doesn't detect any changes either.

If I look in the mbed-os directory I see a .gitignore file. Changing this to .hgignore and then running:
`mbed status`
Give the following:
```
[mbed] Working path "/home/zanev/mbed/AFDAU_0_01" (program)
[mbed] Status for "mbed-os":
 D .gitignore
?? .hgignore
?? UNITTESTS/app/
?? UNITTESTS/bits.cpp
?? UNITTESTS/build/
```
However it isn't detecting the actual unit test .cpp & cmake files.

theotherjimmy, 2019-02-21:
@ZaneV That's because you have made change to mbed-os, which is a different repo (and uses git) from your project.

ZaneV, 2019-02-21:
@theotherjimmy okay so that's what I am concerned about... How do I keep my unit test code apart of my repo and not modify the mbed-os repo if all the unit testing code has to live inside the mbed-os/UNIT TEST folder?

theotherjimmy, 2019-02-21:
@ZaneV I don't think that the unittest mechanism from mbed-os is designed to be used outside the repo. As these are unit tests, you can use any build system that you like, and are not constrained to use mbed-cli

ZaneV, 2019-02-21:
@theotherjimmy okay that makes sense. I will run cmake with Googletest for my own program. The mbed-cli usage documentation on unittests doesn't seem to spell that out too clearly for us rookies and kind of implies that the unittests feature with mbed-cli would be suitable for using on your own program due to being stored inside mbed-os. But that is my opinion. Thanks for all your help.

theotherjimmy, 2019-02-21:
We'll keep this issue open to clarify the usage of `mbed unittest` (or remove it entirely as it's not a user-facing feature)

ZaneV, 2019-02-21:
Okay thanks, could I recommend not removing the unit test section in the documentation but perhaps have a brief guideline on how unit testing might be implemented using cmake and point the reader to some helpful resources. I think the page has been good for me to get an idea of writing unit tests. Anyway thanks again for all your help, I'll leave it in your hands!

theotherjimmy, 2019-02-21:
Sure, we can have the documentation outline how a user could write unit tests for their app/lib. That's independent of having a "sub-subcommand" for it.