ARMmbed/mbed-cli #858: Supress unnessary warning for missing mbed tools

teetak01, 2019-02-25:
Mbed-cli complain a lot about missing tools when working with libraries.

For. ex.
```
mbed import mbed-trace
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "/home/teetak01/devel" (directory)
[mbed] Importing program "mbed-trace" from "ssh://git@github.com/ARMmbed/mbed-trace" at latest revision in the current branch
[mbed] WARNING: Cannot find the mbed tools directory in "/home/teetak01/devel/mbed-trace"
```
I really do not see much benefit from that warning as it comes up with any library deployment, and also with non-Mbed OS projects. It just confuses new users that something might be wrong.

screamerbg, 2019-02-25:
@teetak01 Mbed CLI is an Mbed OS tool, responsible for delivering Mbed OS and interface with the Mbed OS build, test and export tools. Not having the Mbed OS inside a project when the tool by design is expecting Mbed OS to be present, is definitely a reason to see warning about.

Regarding library deployment, if you invoke Mbed CLI from the root of the project (where mbed-os dir is), then there will be no warnings as Mbed CLI can "see" that Mbed OS is present.

teetak01, 2019-02-25:
Is there need for it to be that tied to Mbed OS? Nothing in Mbed CLI really should require such coupling, and this could be platform-agnostic tool.

theotherjimmy, 2019-02-25:
@teetak01 Without mbed-os, you can't run the following subcommands:
* `compile`
* `export`
* `test`

Considering that it's name is also *Mbed* CLI, I would not be surprised that a good portion of it's subcommands are tailored for use with mbed.

theotherjimmy, 2019-02-25:
>  It just confuses new users that something might be wrong.

That's because something might be wrong and many of the tutorials that use Mbed CLI will use the sub-commands that may not work.

closed 2019-02-26 by teetak01

loverdeg-ep, 2019-02-28:
@teetak01 
Yes closed issues get ignored.
It is possibly well known at this point, that I agree with the spirit of this issue.

If Pelion client, for example, is going to be imported via ".lib" for a linux based platform in the same way it is for Mbed OS builds, it would be nice to have an officially supported "platform-agnostic" dependency/package management system that doesn't require the Mbed OS mono-repo be cloned.

> Considering that it's name is also _Mbed_ CLI, I would not be surprised that a good portion of it's subcommands are tailored for use with mbed.

@theotherjimmy 
What is "mbed" considered to be these days? :wink: