ARMmbed/mbed-cli #859: Decoding error when adding CMSIS_5  to project using mbed cli

TylerADavis, 2019-02-28:
I am encountering an error when attempting to add CMSIS_5 to a project using the mbed CLI tools. This occurs both when I attempt to add the library to a fresh project (`mbed new` followed by `mbed add`) as well as when I try to add a library that relies on CMSIS_5 (https://github.com/uTensor/uTensor specifically, via `mbed add https://github.com/uTensor/uTensor`).

I managed to resolve this error in the case of importing uTensor by removing the already imported CMSIS_5 files and then doing `mbed update master` to force the importation of the non development version (main branch is `develop`). I only began encountering this error two days ago while I was able to import the library a few days back, which leads me to believe some recent change to the repo caused this break.

The error message I am seeing looks similar to that in #738 , so this might be related. 

I originally reported this issue in the CMSIS_5 repo and they recommended I come here with the issue first.

I have provided some system info below. If there is anything else I could provide to help with reproducing this, please let me know.

Original error message:
```
(mbed_proj) vagrant@vagrant:/vagrant$ mbed new mbed_proj
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "/vagrant" (directory)
[mbed] Creating new program "mbed_proj" (git)
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at branch/tag "latest"
[mbed] Updating reference "mbed-os" -> "https://github.com/ARMmbed/mbed-os/#ecb3c8c837162c73537bd0f3592c6e2a42994045"
(mbed_proj) vagrant@vagrant:/vagrant$ cd mbed_proj/
(mbed_proj) vagrant@vagrant:/vagrant/mbed_proj$ mbed add https://github.com/ARM-software/CMSIS_5.git
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "/vagrant/mbed_proj" (program)
[mbed] Adding library "CMSIS_5" from "https://github.com/ARM-software/CMSIS_5.git" at latest revision in the current branch
[mbed] ERROR: Unknown Error: 'utf-8' codec can't decode byte 0xb5 in position 134: invalid start byte
---
[mbed] WARNING: Using Python 3 with Mbed OS 5.8 and earlier can cause errors with compiling, testing and exporting
---
```

Fix:
```
(mbed_proj) vagrant@vagrant:~/my_uTensor/uTensor/external/CMSIS_5$ mbed update master
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "/home/vagrant/my_uTensor/uTensor/external/CMSIS_5" (library)
[mbed] Program path "/home/vagrant/my_uTensor"
[mbed] Updating library "CMSIS_5" to branch/tag "master"
[mbed] Updating reference "CMSIS_5" -> "https://github.com/ARM-software/CMSIS_5/#cb3fb9a70f3de3670bf5523a588c7f63cec48ff6"
```

System info:
ubuntu 18.04
```
(mbed_proj) vagrant@vagrant:/vagrant/mbed_proj$ mbed --version
1.9.1
```
```
(mbed_proj) vagrant@vagrant:/vagrant/yolo$ python --version
Python 3.6.7
```
```
(mbed_proj) vagrant@vagrant:/vagrant/$ pip list 
absl-py (0.7.0)
appdirs (1.4.3)
asn1ate (0.6.0)
asn1crypto (0.24.0)
astor (0.7.1)
attrs (18.2.0)
beautifulsoup4 (4.6.3)
certifi (2018.11.29)
cffi (1.12.1)
chardet (3.0.4)
Click (7.0)
colorama (0.3.9)
cryptography (2.5)
ecdsa (0.13)
fasteners (0.14.1)
future (0.16.0)
fuzzywuzzy (0.17.0)
gast (0.2.2)
grpcio (1.18.0)
h5py (2.9.0)
icetea (1.0.2)
idna (2.7)
idx2numpy (1.2.2)
intelhex (2.2.1)
intervaltree (2.1.0)
Jinja2 (2.10)
jsonmerge (1.5.2)
jsonschema (2.6.0)
junit-xml (1.8)
Keras-Applications (1.0.7)
Keras-Preprocessing (1.0.9)
lockfile (0.12.2)
Logbook (1.4.3)
lxml (4.3.1)
manifest-tool (1.4.6)
Markdown (3.0.1)
MarkupSafe (1.1.1)
mbed-cli (1.9.1)
mbed-cloud-sdk (2.0.1)
mbed-flasher (0.9.2)
mbed-greentea (1.6.2)
mbed-host-tests (1.5.2)
mbed-ls (1.7.2)
mbed-os-tools (0.0.3)
monotonic (1.5)
numpy (1.16.1)
pip (9.0.1)
pkg-resources (0.0.0)
prettytable (0.7.2)
protobuf (3.6.1)
py (1.8.0)
pyasn1 (0.2.3)
pycparser (2.19)
pyelftools (0.25)
pyocd (0.14.3)
pyparsing (2.3.1)
pyserial (3.4)
pyshark (0.4.2.2)
python-dateutil (2.8.0)
python-dotenv (0.10.1)
pyusb (1.0.2)
PyYAML (4.2b1)
requests (2.20.1)
semver (2.8.1)
setuptools (40.8.0)
six (1.12.0)
sortedcontainers (2.1.0)
tensorboard (1.12.2)
tensorflow (1.12.0)
termcolor (1.1.0)
urllib3 (1.23)
utensor-cgen (0.3.1)
websocket-client (0.55.0)
Werkzeug (0.14.1)
wheel (0.33.1)
yattag (1.11.1)
```
I also reproduced this issue using macOS 10.14.3 using python 3.7.1 and mbed 1.9.1

theotherjimmy, 2019-02-28:
@TylerADavis Great description of the bug.

desowin, 2019-03-01:
This looks like the same issue I reported to mbed-os-tools: https://github.com/ARMmbed/mbed-os-tools/issues/134

PR #866 (2019-03-05): Fix handling of non mbed .lib files in Python 3
This should fix #859.

Side question (to potentially follow up in another PR): @screamerbg reading only 200 characters here seems a bit low. Could we instead read until the first non-empty line?
closed 2019-03-05 by theotherjimmy