ARMmbed/mbed-cli #860: Ability to use configured variables when compiling

joelvonrotz, 2019-03-01:
The Idea of this Enhancement is to be able to use the configured variables (`MBED_OS_DIR`, `ROOT`, etc.) as parameters, for example when compiling (`mbed compile --source MBED_OS_DIR`). Since it's possible to create custom variables (as an example `LINKER `for linking to custom linker files), there should also be the ability to effectively use them with mbed-cli.

It seems currently (on version 1.9.1) it spits out the error `make.py: error: argument --source: ROOT does not exist in the filesystem` when trying to execute `mbed compile --source ROOT` in a project folder.

Example:

`mbed compile -m LPC1768 -t GCC_ARM --source ROOT --source MBED_OS_DIR`

_(also is there a website where all the variables, which mbed-cli uses, are listed?)_

theotherjimmy, 2019-03-01:
@joelvonrotz This is sounding extremely close to using a bash/sh/fish variable, and any proposed syntax would have to not conflict with:
 * Bash/sh/fish/cmd.exe variables
 * real files or directories in the filesystem

I'm curious about where this expectation came from. Does another tool you use do something like this?

joelvonrotz, 2019-03-06:
> @joelvonrotz This is sounding extremely close to using a bash/sh/fish variable, and any proposed syntax would have to not conflict with:
>
>     * Bash/sh/fish/cmd.exe variables
>
>     * real files or [...]

Hi @theotherjimmy 

It could be that I was just really tired on typing all the parameters in and as a result created this issue. 

What I kind of imagined was to write a command like `mbed compile -l LINKER_LPC1768 --source MBED_OS_DIR --source ROOT`, so not the whole path needs to be typed in (for example, define globally all the linker files via `mbed config -G LINKER_XXX` and use them based on their variables names).

You raise a good point with using cmd variables and I might try to create a batch file with variables. I think then the `mbed config` needs to be limited to the mbed-cli-based variables only (don't allow the creation of more variables -> output error) or make it clear in the documentation that custom variables created with `mbed config` have no effect whatsoever (first thought, it would be possible to use them later on in the mbed commands).

bridadan, 2019-03-06:
> I think then the mbed config needs to be limited to the mbed-cli-based variables only (don't allow the creation of more variables -> output error) or make it clear in the documentation that custom [...]

We agree as well, @naveenkaje is taking a look at this so stay tuned!

joelvonrotz, 2019-03-07:
> We agree as well, @naveenkaje is taking a look at this so stay tuned!

joelvonrotz, 2019-03-07:
darn pressed the wrong button

bulislaw, 2019-05-07:
@joelvonrotz we won't be able to address this proposal anytime soon, please consider working on it and contribute the enhancement back to Mbed OS. Closing it for now.

closed 2019-05-07 by bulislaw